Koenkk/zigbee2mqtt #28112: [New device support]: EcoDim Eco-Dim.07 PRO (likely identical to already supported non-pro version)

lwaeijen, 2025-07-29:
### Link

https://www.ecodim.nl/en/eco-dim07-zigbee-pro.html

### Database entry

{"id":27,"type":"Router","ieeeAddr":"0x04e3e5fffea17378","nwkAddr":2014,"manufId":4714,"manufName":"EcoDim BV","powerSource":"Mains (single phase)","modelId":"eco-dim07-Pro-zigbee","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"eco-dim07-Pro-zigbee","manufacturerName":"EcoDim BV","powerSource":142,"zclVersion":8,"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20250219","swBuildId":"1.02"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20250219","swBuildId":"1.02","zclVersion":8,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1753818108781}

### Zigbee2MQTT version

2.5.1

### Comments

The eco-dim.07 dimmer has a "pro" version, which as far as I can tell only differs from the "non-pro" version in terms of the circuitry, and not in any zigbee related functionality.
Unfortunately it is listed as unsupported upon pairing, although most functionality appears to be discovered correctly.

The database entry looks very similar to the one [reported in a previous issue](https://github.com/Koenkk/zigbee2mqtt/issues/27891) for the non-pro version, apart from the modelId (which adds "Pro" to the name), the swBuildId, and the dateCode.
Hopefully adding another fingerprint will be sufficient to also support this "pro" version.

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['eco-dim07-Pro-zigbee'],
    model: 'eco-dim07-Pro-zigbee',
    vendor: 'EcoDim BV',
    description: 'Automatically generated definition',
    extend: [m.light()],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Dimming and switching on/off appears to work correctly. 
Setting the power-on behaviour reports back as unsupported.

Koenkk, 2025-07-30:
Can you try to stop z2m, change all occurences of `eco-dim07-Pro-zigbee` to `eco-dim07-zigbee` and see if it works?

Koenkk, 2025-07-30:
Added, thanks!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) tomorrow and in the next release which is every 1st of the month.

closed 2025-07-30 by Koenkk