Koenkk/zigbee2mqtt #28113: [New device support] Tuya/NovaDigital TS0601 6‑gang switch (_TZE204_wskr3up8)

cbmd0405, 2025-07-30:
### Link

https://www.novadigitalsmart.com.br/produtos/interruptor-inteligente-zigbee-tecla-fisica-4x4-fzb-4-6

### Database entry

{"id":35,"type":"Router","ieeeAddr":"0xa4c13834c56371af","nwkAddr":25503,"manufId":4417,"manufName":"_TZE204_wskr3up8","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE204_wskr3up8","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1753841164068}

### Zigbee2MQTT version

2.5.1

### Comments

Problem

On Zigbee2MQTT 2.5.1 (Docker), the device pairs successfully but is marked as unsupported in the frontend. No entities are created and I cannot control any of the six channels. The device appears under “how_to_add_support” in the UI.

What I’ve tried

I verified that recent Nova Digital models (SYZB‑6W and NTZB‑04‑W‑B) are supported in Zigbee2MQTT ≥2.4.0, but this fingerprint _TZE204_wskr3up8 is not listed in the dev changelog and I could not find an existing issue for it.

I attempted to create a custom external converter based on the TS0601_switch_6_gang definition for _TZE204_g4au0afs. I adjusted the fingerprint to _TZE204_wskr3up8 and exposed endpoints l1–l6 for each channel. With this converter, the device pairs and I can switch only one channel; the other five channels do not respond. Zigbee2MQTT logs repeated bind errors (INVALID_EP) and reporting failures. Power‑on behaviour, backlight mode and state reporting also do not work.

Removing the external converter and using the auto‑generated definition results in no functionality at all.

Request

Could support be added for this fingerprint? The device is sold under the NovaDigital Smart brand in Brazil (product page: [Interruptor Inteligente ZigBee Tecla Fisíca 4x4](https://www.novadigitalsmart.com.br/produtos/interruptor-inteligente-zigbee-tecla-fisica-4x4-fzb-4-6)). I’m happy to test patches or provide additional logs if needed.

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');

const e = exposes.presets;

const definition = {
    // Identifica o dispositivo pelo modelId e pelo manufacturerName observados no database.db
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_wskr3up8']),
    model: 'TS0601_switch_6_gang',
    vendor: 'Nova Digital',
    description: '6‑gang wall switch (4 physical buttons + 2 sockets)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    // Extende o comportamento on/off padrão Tuya
    extend: [tuya.modernExtend.tuyaOnOff()],
    // Opcional: tenta fazer o bind de todos os canais para o coordenador
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(4), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(5), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(6), coordinatorEndpoint, ['genOnOff']);
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
    // Expõe seis endpoints, um para cada botão/tomada
    exposes: [
        e.switch().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.switch().withEndpoint('l3'),
        e.switch().withEndpoint('l4'),
        e.switch().withEndpoint('l5'),
        e.switch().withEndpoint('l6'),
    ],
    // Mapeia todos os canais para o mesmo endpoint físico (1)
    endpoint: (device) => {
        return {l1: 1, l2: 1, l3: 1, l4: 1, l5: 1, l6: 1};
    },
    meta: {
        multiEndpoint: true,
        // Define os datapoints Tuya correspondentes às seis saídas
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
            [5, 'state_l5', tuya.valueConverter.onOff],
            [6, 'state_l6', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

work only de switch number one. every other switchs not responded.

cbmd0405, 2025-07-30:
info 2025-07-29 23:05:56z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sensor banheiro social', payload '{"battery":100,"battery_low":false,"keep_time":60,"linkquality":128,"occupancy":true,"sensitivity":"medium","voltage":2900}'
info 2025-07-29 23:06:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/corredor 1', payload '{"backlight_mode":"OFF","indicator_mode":"off","linkquality":164,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","state_l1":"OFF","state_l2":"ON","state_l3":"OFF"}'
info 2025-07-29 23:06:08z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/corredor 1', payload '{"backlight_mode":"OFF","indicator_mode":"off","linkquality":168,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","state_l1":"OFF","state_l2":"ON","state_l3":"OFF"}'
info 2025-07-29 23:06:09zh:controller: Interview for '0xa4c13834c56371af' started
info 2025-07-29 23:06:09z2m: Device 'banheiro social' joined
info 2025-07-29 23:06:09z2m: Starting interview of 'banheiro social'
info 2025-07-29 23:06:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"banheiro social","ieee_address":"0xa4c13834c56371af"},"type":"device_joined"}'
info 2025-07-29 23:06:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"banheiro social","ieee_address":"0xa4c13834c56371af","status":"started"},"type":"device_interview"}'
info 2025-07-29 23:06:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"banheiro social","ieee_address":"0xa4c13834c56371af"},"type":"device_announce"}'
info 2025-07-29 23:06:10zh:controller:device: Device '0xa4c13834c56371af' is only compliant to revision '21' of the ZigBee specification (current revision: 23).
info 2025-07-29 23:06:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/corredor 1', payload '{"backlight_mode":"OFF","indicator_mode":"off","linkquality":168,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info 2025-07-29 23:06:12z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/corredor 1', payload '{"backlight_mode":"OFF","indicator_mode":"off","linkquality":164,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info 2025-07-29 23:06:14zh:controller: Succesfully interviewed '0xa4c13834c56371af'
info 2025-07-29 23:06:14z2m: Successfully interviewed 'banheiro social', device has successfully been paired
warning 2025-07-29 23:06:14z2m: Device 'banheiro social' with Zigbee model 'TS0601' and manufacturer name '_TZE204_wskr3up8' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
info 2025-07-29 23:06:14z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c13834c56371af/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c13834c56371af"],"manufacturer":"_TZE204_wskr3up8","model":"Automatically generated definition","model_id":"TS0601","name":"banheiro social","via_device":"zigbee2mqtt_bridge_0x08b95ffffed9f614"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"banheiro_social_linkquality","origin":{"name":"Zigbee2MQTT","sw":"2.5.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/banheiro social","unique_id":"0xa4c13834c56371af_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
info 2025-07-29 23:06:14z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Automatically generated definition","exposes":[{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601","options":[],"supports_ota":false,"vendor":"_TZE204_wskr3up8"},"friendly_name":"banheiro social","ieee_address":"0xa4c13834c56371af","status":"successful","supported":false},"type":"device_interview"}'
info 2025-07-29 23:06:24z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/movimento corredor', payload '{"battery":100,"battery_low":false,"keep_time":30,"linkquality":168,"occupancy":true,"sensitivity":"medium","voltage":2900}'
info 2025-07-29 23:06:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/movimento corredor', payload '{"battery":100,"battery_low":false,"keep_time":30,"linkquality":168,"occupancy":false,"sensitivity":"medium","voltage":2900}'
info 2025-07-29 23:06:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/corredor 1', payload '{"backlight_mode":"OFF","indicator_mode":"off","linkquality":164,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
info 2025-07-29 23:06:26z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/corredor 1', payload '{"backlight_mode":"OFF","indicator_mode":"off","linkquality":168,"power_on_behavior_l1":"off","power_on_behavior_l2":"off","power_on_behavior_l3":"off","state_l1":"OFF","state_l2":"OFF","state_l3":"OFF"}'
[... cut ...] 12,"messages_per_sec":0.0072,"leave_count":0,"network_address_changes":0},"0xa4c13850329adbd2":{"messages":3,"messages_per_sec":0.0019,"leave_count":0,"network_address_changes":0},"0xa4c138739dd0940e":{"messages":3,"messages_per_sec":0.0023,"leave_count":0,"network_address_changes":0},"0xa4c1381f778dcf1e":{"messages":1,"messages_per_sec":0.0014,"leave_count":0,"network_address_changes":0},"0xa4c1385bf3edc5d1":{"messages":4,"messages_per_sec":0.0103,"leave_count":0,"network_address_changes":0}}}'

cbmd0405, 2025-08-01:
Man, I honestly have no words to thank you enough!

I've been trying to fix this issue for months with that Zigbee device — tried everything from external converters to complete reinstallation, and it always broke or got unstable.

Your suggestion to edit the database.db and replace the manufName worked perfectly.
As soon as I restarted Zigbee2MQTT, the device was recognized and started working.

Seriously, thank you so much for sharing this fix. You saved my setup and ended a long-standing headache.

Really appreciate it! 🙏👏

bernardesarthur, 2025-08-01:
Hey bro I'm glad that I helped you with. I'll add that device in next release. Enjoy your weekend. Regards 🙏🏼🙏🏼

closed 2025-08-01 by Koenkk