Koenkk/zigbee2mqtt #28114: Same Tuya pir sensor identified as different model

pbu80, 2025-07-30:
### What happened?

I have ordered the below model tested with Z2M detected as [Tuya ZG-204ZL](https://www.zigbee2mqtt.io/devices/ZG-204ZL.html#tuya-zg-204zl), works well with great battery life, it's been a year now.

Second order gets detected as [eWeLink SNZB-03_eWeLink](https://www.zigbee2mqtt.io/devices/SNZB-03_eWeLink.html#ewelink-snzb-03-ewelink), works fine but sucks battery life, I am swapping batteries every month :(

Any idea what happening here?

Product link

[Zigbee Mini PIR Motion Sensor Human Motion Infrared Detector Security Alarm Work With Alexa Google Home Zigbee2MQTT Need Gateway - AliExpress 30](https://www.aliexpress.com/item/1005007425213269.html?spm=a2g0o.order_list.order_list_main.10.7b4718029GavHq)

### What did you expect to happen?

Should detect as [Tuya ZG-204ZL](https://www.zigbee2mqtt.io/devices/ZG-204ZL.html#tuya-zg-204zl) 

### Zigbee2MQTT version

2.5.0

### Adapter firmware version

7.4.3 [GA]

### Adapter

skyconnect with ember dirver 

### Setup

As plain docker container, running on intel NUC Ubuntu 20.04.6 LTS

pbu80, 2025-08-02:
Thanks for the reply @Nerivec, 
I am very new to z2m, I thought it we could simply force it to detect as intended model, looks like need to swap them out.
I wish these Tuya devices be more constant :)

Nerivec, 2025-08-02:
The model is matched to the known capabilities of the device using the information that the device reports (as best as possible anyway).

Note that Ewelink is Sonoff, and _usually_ better than generic Tuya. You may want to check the link quality thing before replacing them.
Definitely avoid Tuya as much as possible, your network will thank you later 😉

closed 2025-08-02 by Nerivec