Koenkk/zigbee2mqtt #28116: [New device support]: Tuya LED Controler TY-X7-ZIGBEE

AlexanderDedy, 2025-07-30:
### Link

https://www.aipintech.com/c/ZigBeeController/304.html

### Database entry

{"id":66,"type":"Router","ieeeAddr":"0xa4c138be9939e934","nwkAddr":32768,"manufId":4417,"manufName":"_TZE204_zajeikkt","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�\u001c0\u0013�\u001c0e�\u001c0\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":74}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1753869343509}

### Zigbee2MQTT version

2.5.1

### Comments

I have tried all sorts of things, but unfortunately without success so far.

### External definition

```shell
I tried that, but unfortunately it didn't work. But it's probably my fault.
```

### What does/doesn't work with the external definition?

Since I didn't manage to do that, nothing works at all.

MaikSly01, 2025-09-05:
Same problem, just can't get it to work. AI doesn't have a solution either. I have a WS2801 light strip on the device. It works flawlessly via Tuya Cloud, but via Zigbee2MQTT, only ON/OFF is possible; everything else doesn't work.

LG Maik

stasha17, 2026-10-03:
Have you found any workaround for this?  I am struggling heavily.  The best I could do was get it turning off/on, and a bit of brightness change.  but otherwise no solution.

Nordranei, 2026-10-04:
I got a SP107E and it looks close, this might help?

import * as tuya from 'zigbee-herdsman-converters/lib/tuya';
import {presets as e} from 'zigbee-herdsman-converters/lib/exposes';
import {logger} from 'zigbee-herdsman-converters/lib/logger';
import {Color} from 'zigbee-herdsman-converters/lib/color';

const toHex = (value) =>
    Math.round(value).toString(16).padStart(4, '0');

const colorConverter = {
    to: (value, meta) => {
        const parsed = Color.fromConverterArg(value);

        const hsv = parsed.hsv ??
            (parsed.rgb ? parsed.rgb.toHSV() : parsed.xy.toHSV());

        const previous = meta.state?.color ?? {};

        const hue = Number(
            hsv.hue ?? previous.hue ?? previous.h ?? 0,
        );

        const saturation = Number(
            hsv.saturation ??
            previous.saturation ??
            previous.s ??
            100,
        );

        if (
            !Number.isFinite(hue) ||
            !Number.isFinite(saturation) ||
            hue < 0 || hue > 360 ||
            saturation < 0 || saturation > 100
        ) {
            throw new Error(
                `Ugyldig farge: ${JSON.stringify(value)}`,
            );
        }

        // Lysstyrken reguleres separat via DP106.
        const payload =
            toHex(hue === 360 ? 0 : hue) +
            toHex(saturation * 10) +
            '03e8';

        logger.warning(
            `SP107E COLOR: ${JSON.stringify(value)} -> ${payload}`,
            'sp107e',
        );

        return payload;
    },

    from: (value) => {
        const text = String(value).toLowerCase();

        if (!/^[0-9a-f]{12}$/.test(text)) {
            return;
        }

        return {
            hue: parseInt(text.slice(0, 4), 16),
            saturation: parseInt(text.slice(4, 8), 16) / 10,
        };
    },
};

const speedConverter = {
    to: (value) => {
        const speed = Number(value);

        if (
            !Number.isInteger(speed) ||
            speed < 1 ||
            speed > 100
        ) {
            throw new Error('Hastighet må være et heltall fra 1 til 100');
        }

        return 101 - speed;
    },

    from: (value) => 101 - Number(value),
};

export default {
    fingerprint: [{
        modelID: 'TS0601',
        manufacturerName: '_TZE204_zajeikkt',
    }],

    model: 'TY-SP107E-Zigbee',
    vendor: 'Tuya',
    description: 'SP107E - colors, brightness, speed, effects and music',

    extend: [
        tuya.modernExtend.tuyaBase({
            dp: true,
            queryOnConfigure: true,
        }),
    ],

    exposes: [
        e.light_colorhs(),

        e.enum('mode', 3, ['solid', 'effects', 'music'])
            .withLabel('Lysmodus')
            .withDescription('Fast lys, effekter eller musikk'),

        e.numeric('brightness_percent', 3)
            .withLabel('Lysstyrke')
            .withUnit('%')
            .withValueMin(10)
            .withValueMax(100)
            .withValueStep(1)
            .withDescription('Lysstyrke via DP106'),

        e.numeric('speed_test', 3)
            .withLabel('Effekthastighet')
            .withValueMin(1)
            .withValueMax(100)
            .withValueStep(1)
            .withDescription('1 = saktest, 100 = raskest'),

        e.numeric('effect', 3)
            .withLabel('Effekt')
            .withValueMin(0)
            .withValueMax(179)
            .withValueStep(1)
            .withDescription('Effektnummer i effects-modus'),

        e.numeric('sensitivity_test', 3)
            .withLabel('Mikrofonfølsomhet')
            .withUnit('%')
            .withValueMin(1)
            .withValueMax(100)
            .withValueStep(1)
            .withDescription('Mikrofonfølsomhet i musikkmodus'),

        e.numeric('music_effect', 3)
            .withLabel('Musikkeffekt')
            .withValueMin(0)
            .withValueMax(21)
            .withValueStep(1)
            .withDescription('Musikkmønster via DP109'),
    ],

    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff],

            [2, 'mode', tuya.valueConverterBasic.lookup({
                mode_0: tuya.enum(0),
                solid: tuya.enum(1),
                effects: tuya.enum(2),
                music: tuya.enum(3),
            })],

            [5, 'color', colorConverter],
            [106, 'brightness_percent', tuya.valueConverter.raw],
            [107, 'speed_test', speedConverter],
            [108, 'effect', tuya.valueConverter.raw],
            [109, 'music_effect', tuya.valueConverter.raw],
            [110, 'sensitivity_test', tuya.valueConverter.raw],
        ],
    },

    fromZigbee: [{
        cluster: 'manuSpecificTuya',
        type: ['commandDataResponse', 'commandDataReport'],
        convert: (model, msg) => {
            logger.warning(
                `SP107E DATA: ${JSON.stringify(msg.data)}`,
                'sp107e',
            );
            return {};
        },
    }],
};