Koenkk/zigbee2mqtt #28127: Danfoss Icon2 24V RT 088U2128 not pairing

AivCho123, 2025-07-31:
### What happened?

I tried to connect Danfoss Icon2 24V RT 088U2128 thermostat to zigbee2mqtt without success

model id: 0x8020

### What did you expect to happen?

I expected it to connect to zigbee2mqtt

### How to reproduce it (minimal and precise)

Just try to connect Danfoss Icon2 24V RT 088U2128 thermostat

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

 20230507

### Adapter

CC2652P

### Setup

Asus tinkerboard 2S, Debian 11, CC2652P module, Danfoss Icon2 24V RT 088U2128

Nerivec, 2025-07-31:
Please provide a `debug` log of the pairing. https://www.zigbee2mqtt.io/guide/configuration/logging.html#debugging

AivCho123, 2025-08-01:
added log

[log.log](https://github.com/user-attachments/files/21542274/log.log)

AivCho123, 2025-08-12:
Can anyone tell me where the device connection logic is written?

closed 2025-10-19 by github-actions[bot]