Koenkk/zigbee2mqtt #28129: Trying to set power outage behavior for TS0505B bulbs doesn't work

ibbnog2, 2025-07-31:
### What happened?

I have EMOS GU10 and Nous P3Z RGB bulbs with the TS0505B chip and wanted them to not turn on when a power outage happens and they turn back on. Based on issue #23000 , I first tried to do it normally with the "Do not disturb" mode, but that of course didn't help with anything. I then tried to do it through the dev console with cluster genOnOff, attribute startUpOnOff and value 0, but when I got to write it returned "Status 'UNSUPPORTED_ATTRIBUTE'".

### What did you expect to happen?

I expected the write to work and the bulb to not turn on when power got back.

### How to reproduce it (minimal and precise)

In the dev console of the bulb, set cluster to genOnOff, attribute to startUpOnOff and value to 0 and press "Write". After that, turn the bulb off and on from the power and see that it still lights up.

### Zigbee2MQTT version

2.5.1-dev (also tried on the normal version)

### Adapter firmware version

Revision: 8.0.2 [GA]

### Adapter

SONOFF Dongle-E EmberZNet

### Setup

Add-on on Home Assistant OS, inside of a Proxmox VM

### Debug log

[2025-07-31 11:09:22] z2m: Publish 'set' 'write' to '0xa4c138e79b46b975' failed: 'Error: ZCL command 0xa4c138e79b46b975/1 genOnOff.write({"startUpOnOff":255}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

bernardesarthur, 2025-07-31:
Even on Tuya Platform this function doesn't work.

Nerivec, 2025-07-31:
This is driven by the device's firmware, if it returns unsupported, there's nothing Z2M can do.
Note: specific variants may support it, while others may not, so, same device model can sometimes support it, sometimes not (most often depending on firmware version).

closed 2025-07-31 by Nerivec