Koenkk/zigbee2mqtt #28142: Z2M does not provide sensors: SZ-SRN12N

TomberWolf, 2025-08-01:
### What happened?

Sensors and settings are detected in the Z2M backend but aren't provided (in Home Assistant for example).

https://www.zigbee2mqtt.io/devices/SZ-SRN12N.html

### Zigbee2MQTT version

2.5.1

### Adapter firmware version

20250325

### Adapter

SMLight SLZB-06

### Setup

Home Assistant as VM
Z2M Addon

Koenkk, 2025-09-28:
This is because the only expose is a composite one, and HA doesn't support these complex expose types. To integrate you can manually send a MQTT message with HA. See [the docs](https://www.zigbee2mqtt.io/devices/SZ-SRN12N.html#warning-composite) on what to send.

Koenkk, 2025-09-29:
@KameDomotics I think ZHA provides default values while Z2M allow users to chose the values (it's a choice basically). Can you make a screenshot on what kind of entities ZHA provides?

KameDomotics, 2025-09-29:
> I took screenshots of the device settings in ZHA before switching from ZHA to Z2M. Hope this helps. 
> 
> [image]

Thank you @TomberWolf 

In the past I had also tried ZHA to see the difference in integration of this siren (for comparison Z2M with ZHA), but I use Z2M because I find it better.

@Koenkk Can the integration of this siren be improved?

Thanks

snazzybean, 2026-02-07:
> This is because the only expose is a composite one, and HA doesn't support these complex expose types. To integrate you can manually send a MQTT message with HA. See [the docs](https://www.zigbee2mqtt.io/devices/SZ-SRN12N.html#warning-composite) on what to send.

@Koenkk  Maybe I have a solution there so that we get the fields displayed in HA. I was actually working on https://github.com/Koenkk/zigbee2mqtt/issues/30583, since I also got the siren (SIRZB-111), but it doesn't work as expected. For that, I'll try to create a PR in [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters) soon.

But regarding this issue:
It would look something like this (unfortunately in German, but short the fields: sound (mode), volume, duration, start, stop). Locally it works for me so far. What do you think?

[image]

So we could at least get the values correctly in HA and also use them a bit better in automations, without going the manual MQTT route.
I'm not super clued in on all the code and architecture here. But I'll give it my best shot to get the PR merged. I mean.. Currently it works on my machine :D

KameDomotics, 2026-02-08:
> @Koenkk  Maybe I have a solution there so that we get the fields displayed in HA. I was actually working on https://github.com/Koenkk/zigbee2mqtt/issues/30583, since I also got the siren [...]

It would be great to have the siren entities in HA. I'm not a developer, but on ZHA my siren has all the entities, so I think it's possible with Z2M too.

KameDomotics, 2026-02-09:
> @TomberWolf @KameDomotics https://github.com/Koenkk/zigbee2mqtt/pull/31000 is merged. There is a small config example  in the PR for how you can control your siren in the future. And the UI in HA of course. I think that's all we get with this composite.

So in the future, will entities be exposed in HA like they are with ZHA?