Koenkk/zigbee2mqtt #28149: Wrong detection of Tuya Smart knob ZG-101Z/D with 2.6.0

raveit65, 2025-08-01:
### What happened?

A tuya smart knob (ZG-101Z_D) https://www.zigbee2mqtt.io/devices/ZG-101Z_D.html  is now detected as device from vendor hobeian. Which is definitely wrong.
Now it's detected as https://www.zigbee2mqtt.io/devices/ZG-101ZD.html#hobeian-zg-101zd     (404)
And it doesn't work anymore.

[image]
My device doesn't have those options.  Action step size, Action transition time, Action rate
Those options are looking like Tuya ERS-10TZBVK-AA knob.

Correct database entry with zdm 2.5.1:
```
"id":159,"type":"EndDevice","ieeeAddr":"0xa4c138d2d902291b","nwkAddr":51799,"manufId":4742,"manufName":"_TZ3000_gwkzibhs","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_gwkzibhs","powerSource":3,"zclVersion":3,"appVersion":147,"stackVersion":2,"hwVersion":1,"dateCode":"20250612","swBuildId":"0122052017"}},"genOnOff":{"attributes":{"tuyaOperationMode":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002e1e1056","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":147,"stackVersion":2,"hwVersion":1,"dateCode":"20250612","swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1754071548697}
```
I am back on 2.5.1 and need to update again to provide database entry with zdm-2.6.0 

### What did you expect to happen?

A working smart knob.

### How to reproduce it (minimal and precise)

UPdate from zdm from 2.5.1 to 2.6.0

### Zigbee2MQTT version

2.6.0

### Adapter firmware version

20250321

### Adapter

slzb-06

### Setup

As home assistant addon. HA runs as generic-x86-64 on a hp thin client.

### Device `database.db` entry

"id":159,"type":"EndDevice","ieeeAddr":"0xa4c138d2d902291b","nwkAddr":51799,"manufId":4742,"manufName":"_TZ3000_gwkzibhs","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_gwkzibhs","powerSource":3,"zclVersion":3,"appVersion":147,"stackVersion":2,"hwVersion":1,"dateCode":"20250612","swBuildId":"0122052017"}},"genOnOff":{"attributes":{"tuyaOperationMode":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002e1e1056","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":147,"stackVersion":2,"hwVersion":1,"dateCode":"20250612","swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1754071548697}

raveit65, 2025-08-01:
Database entry (_TZ3000_gwkzibhs) with 2.6.0 is
```
{"id":159,"type":"EndDevice","ieeeAddr":"0xa4c138d2d902291b","nwkAddr":51799,"manufId":4742,"manufName":"_TZ3000_gwkzibhs","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TS004F","manufacturerName":"_TZ3000_gwkzibhs","powerSource":3,"zclVersion":3,"appVersion":147,"stackVersion":2,"hwVersion":1,"dateCode":"20250612","swBuildId":"0122052017"}},"genOnOff":{"attributes":{"tuyaOperationMode":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":29}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002e1e1056","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":147,"stackVersion":2,"hwVersion":1,"dateCode":"20250612","swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1754074142667}
```
When updating second time to 2.6.0 the device is working again for some reasons. But having options Action step size, Action transition time, Action rate is wrong.

bernardesarthur, 2025-08-01:
I'll check it out if a can fix this.

raveit65, 2025-08-04:
With zdm 2.6.0 i can select 2 modes, (command and event)
Actions with event mode are the same as with 2.5.1.
- single/toggle
- double
- hold
- rotate_right
- rotate_left

I command mode i see standard + some advanced actions. Same like the Tuya ERS-10TZBVK-AA knob. I own this device too.
- rotate right/brightness_step_up (works)
- rotate left/brightness_step_down (works)
- hold/hue_move (untested)
- hold/hue_stop (untested)
- hold/rotate_right/color_temperature_step_up (untested)
- hold/rotate_left/color_temperature_step_down (untested)

Most of the new actions i can't test or verify to work because i have no idea to write an automation for this. Or maybe i need some special light devices.
But brightness_step_up/down is working. So i think only the vendor name, device picture and model name is wrong with zdm-2.6.0. Tuya writes it with a slash on packaging (ZG-101Z/D). 
About tab with 2.6.0
[image]

I hope i could clarify the issue and i am happy to provide more information, also for Tuya ERS-10TZBVK-AA knob when requested.

raveit65, 2025-08-14:
@ShaunLWM 
It's fixed in z2m edge version.
See https://github.com/Koenkk/zigbee-herdsman-converters/pull/9727#issuecomment-3152049061

Koenkk, 2025-09-02:
Is this now working well? Seems this introduced https://github.com/Koenkk/zigbee2mqtt/issues/28461

raveit65, 2025-09-16:
@Koenkk 
I did resell this weird devices so i can't do any further tests anymore and i will close the issue. But see my previous posts.

Koenkk, 2025-10-30:
Fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/10562

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.