Koenkk/zigbee2mqtt #28150: [Wrong device]: Zemismart ZM25AZ01 detected as Hiladuo B09M3R35GC

alexp999, 2025-08-01:
### Link

https://www.zemismart.com/products/zm25az01

### Model

ZM25AZ01

### Description

Roller Shade Motor

### Vendor

Zemismart

### Picture (link)

https://ueeshop.ly200-cdn.com/u_file/UPBE/UPBE110/2412/products/21/596cd84209.jpg

### Database entry

{"id":16,"type":"EndDevice","ieeeAddr":"0x6cfd22fffe55b029","nwkAddr":62307,"manufId":4098,"manufName":"_TZE200_9p5xmj5r","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0014��/\u0011","65506":31,"65508":0,"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_9p5xmj5r","powerSource":3,"zclVersion":3,"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1754071962734}

### Notes

Generated external definition:

```
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_9p5xmj5r',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```
Everything works perfect except click control.

bernardesarthur, 2025-08-02:
Did this device work in the previous version?

bernardesarthur, 2025-08-02:
I've been looking for while, even release of May/2025 and that device doesn't changed. We wait if anybody else can help you.

alexp999, 2025-08-04:
> Seems they are sold under multiple appearances (was changed in [Koenkk/zigbee-herdsman-converters#6893](https://github.com/Koenkk/zigbee-herdsman-converters/pull/6893)).
> 
> Is it only a cosmetic issue? (wrong description/picture?)

The step function doesn't seem to work, but everything else works 100%.

jpauoc, 2025-08-06:
This is the database entry:

{"id":62,"type":"EndDevice","ieeeAddr":"0x0c2a6ffffe1951d1","nwkAddr":34932,"manufId":4098,"manufName":"_TZE200_9p5xmj5r","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u001f�#0\u0011","65506":31,"65508":0,"65534":0,"appVersion":72,"modelId":"TS0601","manufacturerName":"_TZE200_9p5xmj5r","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1754340510454}

Device information:

Vendor: A-OK (but I think it is OEM)
Model: AM25-1/25-ES-E-Z
Pic: https://ae-pic-a1.aliexpress-media.com/kf/S1e4ab17b80ab47bbbc2693fcd328707cW.png_960x960.png_.avif