Koenkk/zigbee2mqtt #28154: [ZG-227Z / HOBEIAN] ZG-227Z without display is showed like a sensor with display into zigbee2mqtt interface

Danieldiazi, 2025-08-02:
### What happened?
My item:
I've a ZG-227Z, model HZ-HT-02, is round, without display, 

[image]
[image]

and when I add to ziggbe2mqtt a model with LCD display is showed.

[image]

### What did you expect to happen?

Support this variant and change the image according to the real model.

### How to reproduce it (minimal and precise)

### Debug logs

### Details
os: `#1 SMP PREEMPT_DYNAMIC Debian 6.12.33-1~bpo12+1 (2025-07-09) - 6.12.33+deb12-amd64 - x64`
node: `v22.16.0`
zigbee2mqtt: `2.6.0` (`0bdc3fd1f76aaad71f6231e5b38529e2d2b345e9
`)
zigbee-herdsman: `5.0.4`
zigbee-herdsman-converters: `24.11.0`
adapter: `EmberZNet` `{"build":397,"ezsp":14,"major":8,"minor":0,"patch":2,"revision":"8.0.2 [GA]","special":0,"type":170}`
#### Device
software_build_id: `0122052017`
date_code: `undefined`
endpoints:
```json
{"1":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","msTemperatureMeasurement","msRelativeHumidity","genPowerCfg"],"output":["genIdentify"]},"configured_reportings":[],"scenes":[]}}
```
##### Health
time: `Sat Aug 02 2025 12:30:09 GMT+0200 (hora de verano de Europa central)`
process.uptime_sec: `51005`
```json
{"leave_count":0,"messages":103,"messages_per_sec":0.0021,"network_address_changes":0}
```

Albertowue, 2025-10-06:
Hi,
I have the same sensor and I’m experiencing the same issue.
I found this old issue ([#18158](https://github.com/Koenkk/zigbee2mqtt/issues/18158)) which seems to describe exactly the same problem and was apparently fixed a few years ago.
Could this be related or has something changed recently?

Thanks!

Danieldiazi, 2025-11-01:
> Hi, just checking if there are any updates regarding this issue. 
> I can still provide logs or test fixes if needed. Thanks!
> 
> `{"id":9,"type":"EndDevice","ieeeAddr":"0xa4c138d9c671819f","nwkAddr":58414,"manufId":4742,"manufName":"HOBEIAN","powerSource":"Battery","modelId":"ZG-227Z","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"ZG-227Z","manufacturerName":"HOBEIAN","powerSource":3,"zclVersion":3,"appVersion":147,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2100}},"msRelativeHumidity":{"attributes":{"measuredValue":6980}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":147,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1760648714476}`
> 

Thanks for this info, I hope this helps in order to fix this issue

Danieldiazi, 2026-01-13:
> Hi there! I have the same issue. I have a Tuya ZG-227Z, it's the same as in your photo. If I understand correctly, the ZG-227Z from Tuya is not in the list of supported devices in zigbee2mqtt. Moreover, the ZG-227Z is detected together with the ZG-227ZL as a sensor with an LCD screen([here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts#L11225)). That's the problem, and I think the new Tuya ZG-227Z sensor should be added to the supported devices list([here](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)).

I agree!!

closed 2026-03-23 by github-actions[bot]