Koenkk/zigbee2mqtt #28161: Failed to configure <device>, attempt 2 (Error: Device <id> has no output cluster genScenes

plamenko, 2025-08-02:
### What happened?

After auto-update to zigbee2mqtt 2.6.0-1, i started seeing recurrent
```
Failed to configure 'ZigbeeRemote3', attempt 2 (Error: Device 0x588e81fffeb4f23e has no output cluster genScenes
at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:61:15)
at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:239:31)
at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:27)
at Configure.configure (/app/lib/extension/configure.ts:123:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))
```
on my IKEA TRADFRI remote controls (5 button)
Could be similar to https://github.com/Koenkk/zigbee2mqtt/issues/22458

### What did you expect to happen?

Stuff works without errors as before.

### Zigbee2MQTT version

2.6.0-1

### Adapter firmware version

n/a

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

Add-on on HomeAssistant 2025.4.4 (Raspberry Pi 4)

Nerivec, 2025-08-02:
Did you try doing a "re-interview" on the device (right after waking it up, since it's battery-powered)? _(make sure to not pick the "re-configure" button, that's not the same process)_
That should clear the cache and request the whole dataset from the device again.

danergo, 2025-09-02:
I have a similar if not the same! problem:

```
Sep 02 19:02:47 rpi npm[140625]: [2025-09-02 19:02:47] error:         z2m: Failed to configure 'IKEA Tradfri', attempt 3 (Error: Device 0xXYZ has no output cluster genScenes
Sep 02 19:02:47 rpi npm[140625]:     at getEndpointsWithCluster (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:61:15)
Sep 02 19:02:47 rpi npm[140625]:     at configure (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:226:31)
Sep 02 19:02:47 rpi npm[140625]:     at Object.configure (/opt/zigbee2mqtt/node_modules/.pnpm/zigbee-herdsman-converters@23.2.0/node_modules/zigbee-herdsman-converters/src/index.ts:357:27)
Sep 02 19:02:47 rpi npm[140625]:     at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:124:13)
Sep 02 19:02:47 rpi npm[140625]:     at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:204:17))
```

z2m: Starting Zigbee2MQTT version 2.1.3 (commit #ba337bd3)
z2m: Starting zigbee-herdsman (3.2.7)

What do you suggest?
Cheers