Koenkk/zigbee2mqtt #28174: [New device support]: Philips-White-ambience-Being-pendant HUE

Kapteenivamma, 2025-08-05:
### Link

https://www.amazon.co.uk/Philips-White-ambience-Being-pendant/dp/B09C3MLCGD

### Database entry

import * as philips from 'zigbee-herdsman-converters/lib/philips';  export default {     zigbeeModel: ['915005914701'],     model: '915005914701',     vendor: 'Philips',     description: 'Automatically generated definition',     extend: [philips.m.light({"colorTemp":{"range":[153,454]}})],     meta: {}, };

### Zigbee2MQTT version

2.5.1

### Comments

Hello,

This is exactly the same light as https://www.zigbee2mqtt.io/devices/915005914601.html, only difference is the color of the casing in this lamp is white.
The product code is Philips 915005914701

### External definition

```shell
import * as philips from 'zigbee-herdsman-converters/lib/philips';

export default {
    zigbeeModel: ['915005914701'],
    model: '915005914701',
    vendor: 'Philips',
    description: 'Automatically generated definition',
    extend: [philips.m.light({"colorTemp":{"range":[153,454]}})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

I think verything works, but in is seen as not supported

Koenkk, 2025-08-05:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `915005914701.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2025-08-05 by Koenkk

Koenkk, 2025-08-06:
yes please!