Koenkk/zigbee2mqtt #28177: LEDVANCE AC18654 / 4058075729322 not updating physical state change to Z2M after update

Schnulli87, 2025-08-05:
### What happened?

I recently updated Z2M to the latest stable version and afterwards did an OTA update on my LEDVANCE AC18654.
A few days later, I noticed that manual switching on the device is not reported to Z2M. Switching through Z2M reports the state change. I am pretty sure, that it worked normally before both updates, as I use the switch-on time in an openhab rule.

Should be the same as [#26386] (https://github.com/Koenkk/zigbee2mqtt/issues/26386)
and might be related to [#10403] (https://github.com/Koenkk/zigbee2mqtt/issues/10403)

If it is of interest: The AC18654 is detected as the [outdoor version](https://www.zigbee2mqtt.io/devices/4058075729322.html), but that might only be a cosmetic issue according to some issues here.

### What did you expect to happen?

State change should be reported to Z2M when the device is operated manually.

### How to reproduce it (minimal and precise)

Not sure if needed: Update plug to the latest firmware: 20240710
Press button on the device

### Zigbee2MQTT version

2.5.1 commit: ffdd353e

### Adapter firmware version

20240716

### Adapter

SLZB-06P7

### Setup

Z2M running plain on a Raspberry Pi 4:
Linux mqttmatic 6.1.0-rpi4-rpi-v7 #1 SMP Raspbian 1:6.1.54-1+rpt2 (2023-10-05) armv7l GNU/Linux
Raspbian GNU/Linux 12

### Device `database.db` entry

{"id":12,"type":"Router","ieeeAddr":"0xf0d1b800001c3217","nwkAddr":4152,"manufId":4489,"manufName":"LEDVANCE","powerSource":"Mains (single phase)","modelId":"Plug Value","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"zclVersion":8}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0021a8e806","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":2,"dateCode":"20240710","swBuildId":"031F7310","zclVersion":8,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1754422083090}

### Debug log

No entry in the log when operating the button on the device.

Koenkk, 2025-08-06:
Could you try to reconfigure the device by pressing the yellow refresh button in the frontend?

Schnulli87, 2025-08-06:
Sorry, I forgot to mention it: I had already tried your proposed reconfigure function. I had also removed and re-paired this device. All without any effect. 
Just to make sure, I tested the yellow reconfigure again. It immediately caused the starting and successful popups. Unfortunately also without any change to the situation.

Koenkk, 2025-08-11:
Could you provide the debug log when doing the re-configure?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Koenkk, 2025-08-12:
The reporting is successfully configured but somehow the device still refuses to report, I'm afraid this is a bug in the device firmware.

Schnulli87, 2025-08-12:
Okay. Thank you very much for your attempts to try and fix this!
I guess, I will try to not hit the "Update" button again in the future ;-)
Unfortunately, I have not been able to downgrade the firmware, as Z2M finds the older FW file (LEDVANCE provides them), but starting the downgrade results in something like "invalid file".
So I will just exchange this plug with a different one, as we have one lamp in the house, where the button cannot be pressed anyway.

Just to mark this for the other guy with the same issue #26386
@RocketSience, If you read this: We shouldn't have updated our plugs ;-)