Koenkk/zigbee2mqtt #28179: Mureva EV lonk

enthalpie91, 2025-08-05:
### Link

https://www.se.com/fr/fr/product-range/226303288-prises-de-recharge-mureva-evlink/#products

### Database entry

ieeeadsr : 0x70ac08fffe421106

### Zigbee2MQTT version

2.6

### Comments

The value return for Intensity ares 1000 more the real

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['EVSCKT/OUTLET/1'],
    model: 'EVSCKT/OUTLET/1',
    vendor: 'Schneider Electric',
    description: 'Automatically generated definition',
    extend: [m.onOff({"powerOnBehavior":false}), m.electricityMeter()],
    meta: {},
};
```

### What does/doesn't work with the external definition?

the value of intensity is 1000 more than real. so Energy too

enthalpie91, 2025-08-05:
[image]

enthalpie91, 2025-08-06:
Hi 
the change should be on intensity. The value of amperes ares 1000 more than real , so energy is also wrong , but volts are good
COuld you tell me where i sould put these change ? 
best

Koenkk, 2025-08-06:
See https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-1-extending-the-generated-external-definition

enthalpie91, 2025-08-06:
sorry i restart MQTT but the values ares not save, i still have the old values

schauveau, 2025-09-18:
I just posted a converter that appears to solve the problem in https://github.com/Koenkk/zigbee2mqtt/issues/21457#issuecomment-3309445124 

The problem seems to be that the MUR36014 is using both 'seMetering' and 'haElectricalMeasurement' . This is causing problem in electricityMeter because power exists in both clusters.

enthalpie91, 2025-09-19:
well !! i will try i hope it is not too difficult for a newbee

schauveau, 2025-09-19:
Be aware that any syntax error in the external converter will cause the builtin converter to be reused instead.

schauveau, 2025-10-02:
My converter is now available in yesterday's release 2.6.2
https://github.com/Koenkk/zigbee2mqtt/releases/tag/2.6.2

So far, it works fine for me.
A device reconfiguration may be required using the orange buttons in the Devices page.

closed 2025-12-09 by github-actions[bot]