Koenkk/zigbee2mqtt #28188: OTA Update for new Danfoss Ally firmware failed.

vedatbotuk, 2025-08-06:
### What happened?

OTA update failed.
```
info 2025-08-06 19:22:18z2m: Updating 'Danfoss XXXX' to latest firmware

info 2025-08-06 19:22:23zhc:ota: [0x048727fffe16d625 | eTRV0103] Getting latest image, meta: {"fileName":"1246-0120-00320032.0002_(74C7CF9C).ota","fileVersion":32,"fileSize":464046,"url":"https://raw.githubusercontent.com/Koenkk/zigbee-OTA/master/images/Danfoss/1246-0120-00320032.0002_(74C7CF9C).ota","imageType":288,"manufacturerCode":4678,"sha512":"8d536faa3fe43d469fe9ec4f92ea78cce1874872fb82b4401334a76f23158de62ecdfa1af49604726782ba6119c6c60d8363eabc023c2f0dcff3fe06dee2367f","otaHeaderString":"Tue 06/03/2025 V.0020.0020"}

error 2025-08-06 19:23:37z2m: Update of 'Danfoss XXXX' failed (Update failed with reason: ABORT)
```

### What did you expect to happen?

I want to update to new firmware.

### Zigbee2MQTT version

    2.6.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

EmberZNet

### Setup

Home assistant add-on

Nerivec, 2025-08-06:
The device appears to be rejecting the image. See link above.

AresMaster, 2025-08-08:
I have updated my two Popp 701721 (eT093WRO) devices and encountered no problems.

storm1ng, 2025-08-08:
Mine went through as well (eTRV0103)

vedatbotuk, 2025-08-12:
It was the battery problem. With 75% battery I could update.

closed 2025-08-12 by vedatbotuk