Koenkk/zigbee2mqtt #28189: State of Lights is offline after updating

Wallwerker, 2025-08-06:
### What happened?

I have several devices in my HA system via Zigbee2MQTT.

I never had any problems up to Zigbee2MQTT version 2.4.0-1. But as soon as I update to version 2.5.0-1 or higher, the following problem occurs.

All lights, whether lamps, bulbs or LED strips, are suddenly displayed as ‘offline’ in z2m.
However, all sensors, switches and similar devices remain ‘online’.

All automations that control the lights then fail. The automations can no longer address the lights.

In the z2m frontend, the lights marked as ‘offline’ can be switched on and off!
In HA dashboards, it is not possible to switch the lights because they are displayed as unavailable, i.e. offline.

As soon as I roll back to z2m version 2.4.0-1, all devices are immediately back online and everything works as before.

Restarting Z2M, HA or the system it runs on, does not improve the situation. Even ‘waiting until zigbee catches itself’ does not help.

These are only Philips lights. Unfortunately, I don't currently have any lights from other manufacturers, as I switched from a Hue Hub system to Z2M.
None of the sensors and switches (from Philips or other manufacturers) are affected. Only the light sources (lamps, bulbs, LED strips) are affected.

### What did you expect to happen?

After updating Z2M above 2.4.0-1, all devices that shown online before still are shown as online and existing HA automations still work.

### How to reproduce it (minimal and precise)

If I update above 2.4.0.-1 the problem occurs - doesnt matter what version of Z2M above 2.4.0.-1.
As soon as I reinstall the backup of 2.4.0-1, everything works fine again.
I can do that over and over again on my system.

### Zigbee2MQTT version

2.5.0

### Adapter firmware version

7.4.4 [GA]

### Adapter

ZBT-1 EmberZNet Zigbee 7.4.4.1

### Setup

In HA, running in Proxmox as a VM, on a ASUS ExpertCenter PN43-BBN200MD Intel N200

### Debug log

[image]

Nerivec, 2025-08-06:
[Original thread](https://github.com/Koenkk/zigbee2mqtt/discussions/28145#discussioncomment-13987217)

Can you confirm that when you say `switches and similar devices` you mean devices that are mains-powered (routers)? (your screenshot only shows battery devices as "online", so, just want to be sure it's not something specific to mains-powered)

Can you try updating again, set `log_level` to `debug` and grab a few minutes (post Z2M start) worth of logs so we can check what is happening with these lights? We need to know if the lights just stop communicating with Z2M, or if something is wrong with the availability.

Wallwerker, 2025-08-08:
@Nerivec 

[45df7312_zigbee2mqtt_2025-08-08T17-53-26.066Z.log](https://github.com/user-attachments/files/21689435/45df7312_zigbee2mqtt_2025-08-08T17-53-26.066Z.log)

Nerivec, 2025-08-08:
You have an availability timeout of 0 in your `configuration.yaml` for active devices (routers).
Set it to a proper number, and it should go back to normal.
Example:
```yaml
availability:
  enabled: true
  active:
    timeout: 10
  passive:
    timeout: 1500
```

Note: you can also do that in the frontend.

Wallwerker, 2025-08-15:
Please excuse my late reply. 
That helped! Thank you for your help. I hope you can derive something positive from this for yourselves and all users – even if I seemed to be relatively alone with my problem.
I had set the response time a long time ago due to an automation and had forgotten about it, or rather, hadn't considered it as a source of error. It didn't seem to be a problem with previous Z2M UI updates.

Anyway, thanks again!

closed 2025-08-15 by Nerivec