Koenkk/zigbee2mqtt #28196: Unable to start z2m with ZBDongle-E

kozerskil, 2025-08-07:
### What happened?

I've been using z2m successfuly for quite some time already but I decided to start HA and z2m addon. After that I got issues so decided to upgrade dongle as well. Now, z2m is starting with this log:
```
[17:15:28] INFO: Preparing to start...
[17:15:32] INFO: Socat not enabled
[17:15:34] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-08-07 17:16:02] debug: 	z2m: Removing old log directory '/config/zigbee2mqtt/log/2025-08-07.14-28-11'
[2025-08-07 17:16:02] info: 	z2m: Logging to console, file (filename: log.log)
[2025-08-07 17:16:07] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-08-07 17:16:08] info: 	z2m: Starting Zigbee2MQTT version 2.6.0 (commit #unknown)
[2025-08-07 17:16:09] info: 	z2m: Starting zigbee-herdsman (5.0.4)
[2025-08-07 17:16:09] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":18914,\"extendedPanID\":[63,66,11,84,160,174,144,41],\"channelList\":[25],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221028191818-if00\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
[2025-08-07 17:16:10] debug: 	zh:adapter:discovery: Connected devices: [{"path":"/dev/ttyACM0","manufacturer":"ITEAD","serialNumber":"20221028191818","pnpId":"usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221028191818-if00","vendorId":"1a86","productId":"55d4"}]
[2025-08-07 17:16:10] debug: 	zh:adapter:discovery: Unable to match USB adapter: ember | /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221028191818-if00
[2025-08-07 17:16:11] info: 	zh:ember: Using default stack config.
[2025-08-07 17:16:11] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":18914,"extendedPanID":[63,66,11,84,160,174,144,41],"channelList":[25]},"serialPort":{"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221028191818-if00","adapter":"ember"},"adapter":{"disableLED":false},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-08-07 17:16:11] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-08-07 17:16:11] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-08-07 17:16:11] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-08-07 17:16:11] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-08-07 17:16:11] debug: 	zh:ember:uart:ash: Opening serial port with {"path":"/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20221028191818-if00","baudRate":115200,"rtscts":false,"autoOpen":false,"parity":"none","stopBits":1,"xon":true,"xoff":true}
[2025-08-07 17:16:11] info: 	zh:ember:uart:ash: Serial port opened
[2025-08-07 17:16:11] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-08-07 17:16:11] error: 	zh:ember:uart:ash: Received frame with CRC error
[2025-08-07 17:16:11] debug: 	zh:ember:uart:ash: ---> [FRAME type=RST]
[2025-08-07 17:16:11] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-07 17:16:11] error: 	zh:ember:uart:ash: Received frame with CRC error
[2025-08-07 17:16:11] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 0/5000
[2025-08-07 17:16:12] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 100/5000
[2025-08-07 17:16:12] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 200/5000
[2025-08-07 17:16:12] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 300/5000
[2025-08-07 17:16:12] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 400/5000
[2025-08-07 17:16:12] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 500/5000
[2025-08-07 17:16:12] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 600/5000
[2025-08-07 17:16:12] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 700/5000
[2025-08-07 17:16:12] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 800/5000
[2025-08-07 17:16:12] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 900/5000
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 1000/5000
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=RSTACK]
[2025-08-07 17:16:13] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=0)
[2025-08-07 17:16:13] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-08-07 17:16:13] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-08-07 17:16:13] debug: 	zh:ember:ezsp: ===> [FRAME: ID=0:"VERSION" Seq=0 Len=4]
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=0 frmRx=0](ackRx=0)
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1](ackRx=0 frmTx=1)
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=1 frmNum=0](frmRx=0) Added to rxQueue
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=1)
[2025-08-07 17:16:13] debug: 	zh:ember:ezsp: <=== [FRAME: ID=0:"VERSION" Seq=0 Len=7]
[2025-08-07 17:16:13] debug: 	zh:ember:ezsp: ===> [FRAME: ID=0:"VERSION" Seq=1 Len=6]
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=1 frmRx=1](ackRx=1)
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2](ackRx=1 frmTx=2)
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=2 frmNum=1](frmRx=1) Added to rxQueue
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=2)
[2025-08-07 17:16:13] debug: 	zh:ember:ezsp: <=== [FRAME: ID=0:"VERSION" Seq=1 Len=9]
[2025-08-07 17:16:13] info: 	zh:ember: Adapter EZSP protocol version (14) lower than Host. Switched.
[2025-08-07 17:16:13] debug: 	zh:ember: Adapter info: EZSPVersion=14 StackType=2 StackVersion=32800
[2025-08-07 17:16:13] debug: 	zh:ember:ezsp: ===> [FRAME: ID=170:"GET_VALUE" Seq=2 Len=6]
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=2 frmRx=2](ackRx=2)
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=2 frmTx=3)
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=2](frmRx=2) Added to rxQueue
[2025-08-07 17:16:13] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=3)
[2025-08-07 17:16:13] debug: 	zh:ember:ezsp: <=== [FRAME: ID=170:"GET_VALUE" Seq=2 Len=17]
[2025-08-07 17:16:13] info: 	zh:ember: Adapter version info: {"ezsp":14,"revision":"8.0.2 [GA]","build":397,"major":8,"minor":0,"patch":2,"special":0,"type":170}
[2025-08-07 17:16:13] debug: 	zh:ember:ezsp: ===> [FRAME: ID=83:"SET_CONFIGURATION_VALUE" Seq=3 Len=8]
[... 143 lines cut ...]
[2025-08-07 17:16:14] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-07 17:16:14] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6](ackRx=5 frmTx=6)
[2025-08-07 17:16:14] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=6 frmNum=5](frmRx=5) Added to rxQueue
[2025-08-07 17:16:14] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=6)
[2025-08-07 17:16:14] debug: 	zh:ember:ezsp: <=== [FRAME: ID=3:"GET_EXTENDED_VALUE" Seq=21 Len=12]
[2025-08-07 17:16:14] debug: 	zh:ember:ezsp: ===> [FRAME: ID=2:"ADD_ENDPOINT" Seq=22 Len=17]
[2025-08-07 17:16:14] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=6 frmRx=6](ackRx=6)
[17:18:01] INFO: Preparing to start...
```

### Zigbee2MQTT version

2.6.0

### Adapter firmware version

v2024.6.2-update7/sonoff_zbdonglee_zigbee_ncp_8.0.2.0_115200_sw_flow.gbl

### Adapter

ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

### Setup

HassOS on RPi

Nerivec, 2025-08-08:
Can you try doing a full reboot of HA (the red button "Reboot system")?

senna1992, 2025-08-09:
Did you try setting "/dev/ttyACM0" as adapter?

wigledav, 2025-08-09:
I have something similar. Z2M just stopped working whilst I was away after months of no issues. Problem occurred on 2.4.0 (and also 2.6.0). Also tried to update Sonoff ZBDongle-E from 7.4.4 to 8.02 firmware. Same start-up failure.

Running HA Yellow on Pi5. HAOS:
Core
2025.6.1 (also tried on 2025.8)
Supervisor
2025.07.3
Operating System
16.0
Frontend
20250531.3

Z2M will not start and reports the following log snippet:

[2025-08-09 14:30:04] warning: 	zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac2020682af7e]
[2025-08-09 14:30:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=ERROR]
[2025-08-09 14:30:04] error: 	zh:ember:uart:ash: Received ERROR from adapter, with code=RESET_ASSERT.
[2025-08-09 14:30:04] error: 	zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-08-09 14:30:04] error: 	zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-08-09 14:30:04] error: 	zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-08-09 14:30:04] debug: 	zh:controller: Adapter disconnected
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Total frames: RX=87, TX=123
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Cancelled   : RX=1, TX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   DATA frames : RX=85, TX=36
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   DATA bytes  : RX=937, TX=397
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   ACK frames  : RX=0, TX=86
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   CRC errors      : RX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Comm errors     : RX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Length < minimum: RX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Length > maximum: RX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Bad controls    : RX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Retry dupes     : RX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   Out of sequence : RX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash: Port closed.
[2025-08-09 14:30:04] info: 	zh:ember:uart:ash: ======== ASH stopped ========
[2025-08-09 14:30:04] info: 	zh:ember:ezsp: ======== EZSP stopped ========
[2025-08-09 14:30:04] info: 	zh:ember: ======== Ember Adapter Stopped ========
[2025-08-09 14:30:04] error: 	z2m: Adapter disconnected, stopping
[2025-08-09 14:30:04] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
[2025-08-09 14:30:04] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-08-09 14:30:04] info: 	z2m: Disconnecting from MQTT server
[2025-08-09 14:30:04] info: 	z2m: Stopping zigbee-herdsman...
[2025-08-09 14:30:04] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-08-09 14:30:04] info: 	z2m: Stopped zigbee-herdsman
[2025-08-09 14:30:04] info: 	z2m: Stopped Zigbee2MQTT

Nerivec, 2025-08-09:
@wigledav this does not appear to be the same issue (original post, Z2M starts fine, then the logs break and the container seems to start again, which points to HA troubles).
`RESET_ASSERT` means the firmware on the device is giving up because of an unrecoverable error. Could you provide the logs right before the start of the ones you posted so we can see what happened right before it crashed?

wigledav, 2025-08-09:
Hi Nerivec. Thanks for having a look. I also use ZHA with the inbuilt Home Assistant Yellow coordinator and have tried with that disabled to ensure there should not be a data crash as the Z2M logs do refer to a router error address conflict for '0'.
Here are some more of the logs:

debug: 	z2m:mqtt: Received MQTT message on 'homeassistant/light/0x0017880108bebfaf/light/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/Landing Light/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","brightness":true,"brightness_scale":254,"command_topic":"zigbee2mqtt/Landing Light/set","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0x0017880108bebfaf"],"manufacturer":"Philips","model":"Hue white A60 bulb E27 bluetooth","model_id":"8718699673147","name":"Landing Light","sw_version":"1.122.2","via_device":"zigbee2mqtt_bridge_0x08b95ffffeb30de8"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","candle","finish_effect","stop_effect","stop_hue_effect"],"name":null,"object_id":"landing_light","origin":{"name":"Zigbee2MQTT","sw":"2.4.0","url":"https://www.zigbee2mqtt.io"},"schema":"json","state_topic":"zigbee2mqtt/Landing Light","supported_color_modes":["brightness"],"unique_id":"0x0017880108bebfaf_light_zigbee2mqtt"}'
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=3 frmTx=3)
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=7](frmRx=7) Added to rxQueue
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=3)
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=3 frmTx=3)
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=0](frmRx=0) Added to rxQueue
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=3)
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=3 frmTx=3)
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=1](frmRx=1) Added to rxQueue
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=2](ackRx=3)
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=3 frmTx=3)
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=2](frmRx=2) Added to rxQueue
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=3](ackRx=3)
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3](ackRx=3 frmTx=3)
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=3 frmNum=3](frmRx=3) Added to rxQueue
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=3)
[2025-08-09 16:41:04] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=36:"TRUST_CENTER_JOIN_HANDLER" Seq=34 Len=19]
[2025-08-09 16:41:04] debug: 	zh:ember:ezsp: ezspTrustCenterJoinHandler: newNodeId=46632 newNodeEui64=0xd44867fffed3c19e status=STANDARD_SECURITY_SECURED_REJOIN policyDecision=NO_ACTION parentOfNewNodeId=21323
[2025-08-09 16:41:04] debug: 	zh:controller: Device '0xd44867fffed3c19e' joined
[2025-08-09 16:41:04] debug: 	zh:controller: Device '0xd44867fffed3c19e' accepted by handler
[2025-08-09 16:41:04] debug: 	z2m: Device 'Shed Door Sensor' reconnected
[2025-08-09 16:41:04] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Shed Door Sensor/availability', payload '{"state":"online"}'
[2025-08-09 16:41:04] info: 	zh:controller: Interview for '0xd44867fffed3c19e' started
[2025-08-09 16:41:04] debug: 	zh:controller:device: Interview - start device '0xd44867fffed3c19e'
[2025-08-09 16:41:04] debug: 	zh:ember: ~~~> [ZDO NODE_DESCRIPTOR_REQUEST UNICAST to=0xd44867fffed3c19e:46632 messageTag=1 payload=0128b6]
[2025-08-09 16:41:04] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=35 Len=25]
[2025-08-09 16:41:04] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=3 frmRx=4](ackRx=3)
[2025-08-09 16:41:04] info: 	z2m: Starting interview of 'Shed Door Sensor'
[2025-08-09 16:41:04] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Shed Door Sensor","ieee_address":"0xd44867fffed3c19e","status":"started"},"type":"device_interview"}'
[2025-08-09 16:41:04] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=36:"TRUST_CENTER_JOIN_HANDLER" Seq=34 Len=19]
[2025-08-09 16:41:04] debug: 	zh:ember:ezsp: ezspTrustCenterJoinHandler: newNodeId=46632 newNodeEui64=0xd44867fffed3c19e status=STANDARD_SECURITY_SECURED_REJOIN policyDecision=NO_ACTION parentOfNewNodeId=21323
[2025-08-09 16:41:04] debug: 	zh:controller: Device '0xd44867fffed3c19e' joined
[2025-08-09 16:41:04] debug: 	zh:controller: Device '0xd44867fffed3c19e' accepted by handler
[2025-08-09 16:41:04] debug: 	zh:controller: Not interviewing '0xd44867fffed3c19e', interviewState=IN_PROGRESS'
[2025-08-09 16:41:04] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=196:"INCOMING_NETWORK_STATUS_HANDLER" Seq=34 Len=8]
[... cut ...]
[2025-08-09 16:41:06] debug: 	z2m: Saving state to file /config/zigbee2mqtt/state.json
[2025-08-09 16:41:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-08-09 16:41:06] info: 	z2m: Disconnecting from MQTT server
[2025-08-09 16:41:06] info: 	z2m: Stopping zigbee-herdsman...
[2025-08-09 16:41:06] debug: 	zh:controller:database: Writing database to '/config/zigbee2mqtt/database.db'
[2025-08-09 16:41:06] info: 	z2m: Stopped zigbee-herdsman
[2025-08-09 16:41:06] info: 	z2m: Stopped Zigbee2MQTT