Koenkk/zigbee2mqtt #28197: SLZB-MR3: repeated crashes for the EFR32MG24 radio

9shearer, 2025-08-07:
### What happened?

A few days back, I got a new SLZB-MR3 coordinator for testing. I managed to set it up today and noticed the EFR32MG24 instance crashes repeatedly.
If anyone is not aware: the SLZB-MR3 has two Zigbee radios, one based on a EFR32MG24 chip, and the other on a CC2674P10 chip. 
This seems to happen on the hour or approximately so - had one at 16:17, and before that at 12:34, 13:34, 14:35 (then I took a break to update z2m).

### What did you expect to happen?

Coordinator is stable and doesn't restart

### How to reproduce it (minimal and precise)

See test setup - minimal (z2m docker instance, coordinator, one endpoint - wait for the issue to happen on the EFR radio)

### Zigbee2MQTT version

2.6.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

SLZB-MR3

### Setup

Test setup:
- SLZB-MR3 updated to the latest stable firmware for the OS and the two Zigbee radios via the SLZB web interface
- both Zigbee radios in Coordinator mode, on different channels (and different from the rest of the ZB stuff in my home)
- two separate zigbee2mqtt instances (one for each radio) via docker, both on the same virtual machine 
- SLZB-MR3 connected via Ethernet (and powered via POE) to the same switch into which the VM hosting the zigbee2mqtt containers is connected
- no errors whatsoever on the second radio (the one with the CC2674P10 chip)
- zigbee2mqtt latest stable version (2.6.0)
- only one Zigbee device connected to each coordinator (the famous Ikea Parasoll window sensors)
- no HomeAssistant connection

### Debug log

Error messages (sorry for the bad formatting, it's from the web interface):
```
Error 2025-08-07 17:17:50zh:ember:ezsp: ERROR Transaction failure; status=ASH_ERROR_TIMEOUTS. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=34 Len=10].
Error 2025-08-07 17:17:50zh:ember: Adapter fatal error: ASH_ERROR_TIMEOUTS
Info 2025-08-07 17:17:50zh:ember:uart:ash: ASH COUNTERS since last clear:
Info 2025-08-07 17:17:50zh:ember:uart:ash: Total frames: RX=41, TX=78
Info 2025-08-07 17:17:50zh:ember:uart:ash: Cancelled : RX=0, TX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: DATA frames : RX=40, TX=36
Info 2025-08-07 17:17:50zh:ember:uart:ash: DATA bytes : RX=558, TX=375
Info 2025-08-07 17:17:50zh:ember:uart:ash: Retry frames: RX=0, TX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: ACK frames : RX=0, TX=41
Info 2025-08-07 17:17:50zh:ember:uart:ash: NAK frames : RX=0, TX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: nRdy frames : RX=0, TX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: CRC errors : RX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: Comm errors : RX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: Length < minimum: RX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: Length > maximum: RX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: Bad controls : RX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: Bad lengths : RX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: Bad ACK numbers : RX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: Out of buffers : RX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: Retry dupes : RX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: Out of sequence : RX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: ACK timeouts : RX=0
Info 2025-08-07 17:17:50zh:ember:uart:ash: ======== ASH stopped ========
Info 2025-08-07 17:17:50zh:ember:ezsp: ======== EZSP stopped ========
Info 2025-08-07 17:17:50zh:ember: ======== Ember Adapter Stopped ========
Error 2025-08-07 17:17:50z2m: Adapter disconnected, stopping
```

9shearer, 2025-08-09:
The mystery thickens. While the behavior continues, it appears I have different error messages in the frontend from the log file.
Frontend:
[image]

Log file: 
[image]

The overall behavior is still the same. The Z2M container gets restarted on exactly 1 hour and 20 seconds (+/- 1 second, I guess that's a rounding issue), but only for the EFR instance. The CC-based one is steady all along.

9shearer, 2025-08-10:
Looks like the firmware update fixed it. No restarts for ~6 hours.

For context: 
- I had the factory firmware installed, which was listed as 2.9.6
- I have now installed the latest firmware available in the SLZB-OS interface, listed as 2.9.7. It now shows as the "Installed" firmware (green in the Check for firmware popup).

The funny part: the official announcement popup (that you get when opening SLZB-OS web interface) says the following:

[image]

9shearer, 2025-08-12:
A few thoughts:
- make sure your VMWare host (the Win11 machine) never goes to sleep or hibernate and that "allow the system to turn off this device to save power" or whatever it is called (somewhere in Device Manager - device - Properties - Advanced? - Power Saving?) is disabled for the network adapter
- I would suggest using IP addresses instead of mDNS entries (e.g. tcp://10.1.2.3:6638 instead of tcp://slzb-06.local:6638)

My setup is quite basic/default in terms of configuration. The only specific additional setting in z2m is for availability (plus SSL certs etc. for the web interface); on the adapter side, I am on the same firmware levels now, and I have these set up:

[image]

Mrsash2020, 2025-08-12:
Thanks for that. I changed the power settings and the IP addresses to see but been using it the same way for slzb-06 as well. The PC never hibernates or sleeps. The settings in the adapter interface I am going to try. Was also considering changing the port back to 7638 which I am not sure is going to do anything either.

Are you still getting the crashes?

9shearer, 2025-08-13:
> Thanks for that. I changed the power settings and the IP addresses to see but been using it the same way for slzb-06 as well. The PC never hibernates or sleeps. The settings in the adapter interface [...]

I don't think the port has any impact, but you can of course change it.
No crashes since updating the SLZB-OS to 2.9.7. At the moment, I don't have any devices connected to that radio though, but the container has been up and running for two days now.

9shearer, 2025-08-14:
I'm closing this as the issue has now been resolved. Solution - firmware update.