Koenkk/zigbee2mqtt #28200: [New device support]:

adilson0888, 2025-08-07:
### Link

https://www.zemismart.com/products/zms-206us?srsltid=AfmBOoq-6cXhedS2o1PaeT2hyQTiUEKQUIvCrnqvGKePf5LwWSmCjftP&VariantsId=17623

### Database entry

{"id":68,"type":"Router","ieeeAddr":"0xa4c1384eb6818d34","nwkAddr":41278,"manufId":4417,"manufName":"_TZE284_y4jqpry8","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0011\u0000\u0000\u0000i\u0012\u0000\u0000\u0000i\u0012\u0000\u0000\u0000i\u0013\u0000\u0000\u0000i\u0013\u0000\u0000\u0000i\u0014\u0000\u0000\u0000i\u0014\u0000\u0000\u0000i\u0014\u0000\u0000\u0000i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":78,"manufacturerName":"_TZE284_y4jqpry8","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1754593825156}

### Zigbee2MQTT version

2.6.0

### Comments

Hello! I believe it can be another fingerprint variation for zms-206us-4 (4 gang) device. 

I successfully changed the vendor from '_TZE284_y4jqpry8'  to '_TZE284_wwaeqnrf' in the database.db and it worked as expected.

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_y4jqpry8',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Not supported

ubeaut, 2025-08-08:
Hi,
I have the same switch as you and I changed the entries in the database same as you did and I can now see the switches and the exposes etc but I cannot get the switches to work. Did you do anything extra to make it work?
I am getting no converter available for get state and a whole lot of other  . I am using the docker version of zigbee2mqtt and all my other devices are working.
I am at a loss at what to do as I am not experienced in this sort of stuff. I have tried all day to get this far.

Thanks

Koenkk, 2025-08-08:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/9761

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-08-08 by Koenkk