Koenkk/zigbee2mqtt #28214: Error when pairing Inovelli Dimmer Switch

Sandriell, 2025-08-09:
### What happened?

Throwing the following error when pairing an Inovelli Dimmer Switch

`2025-08-08 22:58:41z2m: Failed to configure '0x8c8b48fffe4ce65e', attempt 1 (Error: ZCL command 0x8c8b48fffe4ce65e/1 manuSpecificInovelli.read(["defaultLed6ColorWhenOn","defaultLed6ColorWhenOff","defaultLed6IntensityWhenOn","defaultLed6IntensityWhenOff","defaultLed7ColorWhenOn","defaultLed7ColorWhenOff","defaultLed7IntensityWhenOn","defaultLed7IntensityWhenOff","fanTimerMode","doubleTapClearNotifications"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"manufacturerCode":4655,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:372:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1005:26) at processTicksAndRejections (node:internal/process/task_queues:105:5) at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:481:29) at chunkedRead (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/devices/inovelli.ts:2243:9) at configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/devices/inovelli.ts:2344:13) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:21) at Configure.configure (/app/lib/extension/configure.ts:123:13) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))`

### What did you expect to happen?

No error.

### How to reproduce it (minimal and precise)

Pair the switch

### Zigbee2MQTT version

2.6.0

### Adapter firmware version

20250325

### Adapter

SMLIGHT SLZB-06 Ethernet POE Adapter

### Setup

Add-on on HomeAssistant OS in VM

Sandriell, 2025-08-09:
After further reading [here](https://community.inovelli.com/t/vzm31-sn-reporting-10x-actual-power/19773/13), it seems this is related to https://github.com/Koenkk/zigbee-herdsman-converters/pull/9756, so will have to wait for that to get pushed next month.

Koenkk, 2025-08-09:
Fixed in the dev branch already.

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-08-09 by Koenkk