Koenkk/zigbee2mqtt #28215: [New device support]:  TS0601 _TZE200_h3l0cjqb Tuya blinds motor

stopeject, 2025-08-09:
### Link

https://www.a-okmotor.com/am25-dc-intelligent-zigbee-tubular-motor_p13.html

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0x6cfd22fffe7f54f2","nwkAddr":42593,"manufId":4098,"manufName":"_TZE200_h3l0cjqb","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"�c)0\u0013�c)0\u0012�c)0\u0012","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_h3l0cjqb","stackVersion":0,"dateCode":"","appVersion":72,"powerSource":1,"zclVersion":3,"hwVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1754718591760}

### Zigbee2MQTT version

2.6.0

### Comments

Spent hours with different AIs attempting to build YAML or JS converters with no results. No controls are exposed.
Adding a couple of files for reference. 
[conv.js.txt](https://github.com/user-attachments/files/21696332/conv.js.txt)

[devices.txt](https://github.com/user-attachments/files/21696336/devices.txt)

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_h3l0cjqb',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

controls are not exposed

Koenkk, 2025-08-09:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE200_h3l0cjqb` to `_TZE200_7shyddj3`
- Start Z2M and see if the device now works

eduseiden, 2025-08-09:
Hi @stopeject , probably it won't work fine in your device model, but I use this external converter for tuya blinds.

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definitions = [
  {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_9vpe3fl1'},
    ],
    whiteLabel: [
        tuya.whitelabel('Zemismart', 'ZM25RX-08/30', 'Tubular motor', ['_TZE200_bv1jcqqu', '_TZE200_7eue9vhc']),
    ],
    model: 'TS0601_cover_8',
    vendor: 'TuYa',
    description: 'Cover motor',
    onEvent: tuya.onEvent(),
    configure: tuya.configureMagicPacket,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    options: [exposes.options.invert_cover()],
    exposes: [
        //e.text('work_state', ea.STATE),
        e.cover_position().setAccess('position', ea.STATE_SET),
        //e.battery(),
        // it's also has `UPPER`/`LOWER` command that can be used to move motor over set limit, but it's not clear why :)
        e.enum('program', ea.SET, ['SET BOTTOM', 'SET UPPER', 'RESET', /* 'UPPER', 'LOWER', */]).withDescription('Set the upper/bottom limit'),
        e.enum('motor_direction', ea.STATE_SET, ['NORMAL', 'REVERSED']).withDescription('Motor direction'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
            [2, 'position', tuya.valueConverter.coverPosition],
            [3, 'position', tuya.valueConverter.coverPosition],
            [5, 'motor_direction', tuya.valueConverterBasic.lookup({'NORMAL': tuya.enum(0), 'REVERSED': tuya.enum(1)})],
            [7, 'work_state', tuya.valueConverterBasic.lookup({'standby': tuya.enum(0), 'success': tuya.enum(1), 'learning': tuya.enum(2)})],
            //[13, 'battery', tuya.valueConverter.raw],
            [101, 'program', tuya.valueConverterBasic.lookup({'SET BOTTOM': tuya.enum(0), 'SET UPPER': tuya.enum(1), 'RESET': tuya.enum(4), 'LOWER': tuya.enum(2), 'UPPER': tuya.enum(3)})],
        ],
    },
  },
]

module.exports = definitions;
`

stopeject, 2025-08-09:
@Koenkk , thank you so much, it was like a magic. The controls and reverse switch work, the position is indicated correctly, only the motor speed slider doesn't make a difference. This function is supported because I could change the speed when paired with Tuya. This is a less important control, I can use the blind without it.
@eduseiden, thank you for providing the converter, I'll try it later today. Should I revert the db entry to the original or does it even matter?

eduseiden, 2025-08-09:
> [@Koenkk](https://github.com/Koenkk) , thank you so much, it was like a magic. The controls and reverse switch work, the position is indicated correctly, only the motor speed slider doesn't make a [...]

You have to edit the converter to provide the correct fingerprint. In this part of the code:

fingerprint: [
{modelID: 'TS0601', manufacturerName: '_TZE200_9vpe3fl1'},

closed 2025-10-16 by github-actions[bot]