Koenkk/zigbee2mqtt #28240: [New device support]: new Device Moes Smart Curtain Switch

ha-bnst, 2025-08-12:
### Link

https://www.aliexpress.com/item/1005009362216032.html?spm=a2g0o.productlist.main.7.eda9KO4jKO4jV8&algo_pvid=a3ca57f9-7255-4346-a517-a94d90d0c737&algo_exp_id=a3ca57f9-7255-4346-a517-a94d90d0c737-6&pdp_ext_f=%7B%22order%22%3A%2211%22%2C%22eval%22%3A%221%22%7D&pdp_npi=6%40dis%21AUD%2153.17%2124.99%21%21%21245.36%21115.32%21%40210330dd17549687281246999e30e9%2112000048891712948%21sea%21AU%21937831196%21X%211%210%21n_tag%3A-29919%3Bm03_new_user%3A-29895&curPageLogUid=mDDt1MABU90D&utparam-url=scene%3Asearch%7Cquery_from%3A

### Database entry

{"id":35,"type":"Router","ieeeAddr":"0xa4c13866822efcae","nwkAddr":62551,"manufId":4417,"manufName":"_TZE284_b7kbnl6q","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��-0f��-0\u0012","65506":56,"65508":0,"stackVersion":0,"dateCode":"","appVersion":78}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":78,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1754968219556}

### Zigbee2MQTT version

2.6.0-1

### Comments

I have paired this switch but it is currently unsupported. Can you please assist.

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE284_b7kbnl6q',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

Nothing works

Koenkk, 2025-08-12:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE284_b7kbnl6q` to `_TZE204_lh3arisb`
- Start Z2M and see if the device now works

ha-bnst, 2025-08-12:
Hi Koenkk, Yes that did work but I found another device that seems a better fit _TZE200_nhyj64w2. This has worked so far and includes settings for calibration.