Koenkk/zigbee2mqtt #28243: Z2M HA add-on fails to connect to SLZB-MR3 (fresh setup)

Horgix, 2025-08-12:
### What happened?

Setting up a totally fresh Home Assistant setup (so **no ZHA** or anything existing prior), and and Zigbee2MQTT fails to start due to not being able to connect to the SMLIGHT SLZB-MR3 adapter.

### What did you expect to happen?

I expected Zigbee2MQTT to start find and connect to the adapter to connect Zigbee devices.

### How to reproduce it (minimal and precise)

- Start a fresh Home Assistant setup on x86_64
- Connect a SMLIGHT SLZB-MR3 in USB (with up-to-dates firmware)
- Setup Mosquitto as an HA add-on
- Install Z2M as an HA add-on
- Configure Z2M connection to the SLZB-MR3 through the onboarding page
- Check logs to see it crash

### Zigbee2MQTT version

`2.6.0-1` (didn't see anything in the `dev` branch changelog that could be related to the issue)

### Adapter firmware version

`20240716`

### Adapter

SMLIGHT SLZB-MR3

### Setup

- Adapter: SMLIGHT SLZB-MR3
    - trying to use its **`CC2674P10`** chip for Zigbee - while keeping the `EFR32MG24` for Matter-over-Thread
    - All firmware up-to-date (Core `v2.9.7`, CC2674P10 `20240716`)
    - Yes, I'm using the Zigbee **coordinator** firmware, _not_ the _router_ one
    - Connected in **USB** mode with a 4m long USB-A to USB-C - _not_ over LAN
    - Connect to a USB-A **3.1** port; I believe a USB 2 is to be favored but I got no such port on the hardware 😕 Might the issue be  due to this?
- Home Assistant **OS**
    - Home Assistant Core `2025.8.0`
    - Running on a x86_64 machine (HP EliteDesk 800 G4 fwiw)
- Zigbee2MQTT installed as an **add-on** - version `2.6.0-1`
    - Trying to use Zigbee channel `25` or `20`

On Zigbee2MQTT config, essentially (configured through the "onboarding" UI at add-on startup):

```yaml
serial:
  rtscts: false
  port: /dev/ttyUSB0
  adapter: zstack
  baudrate: 115200
```

Things I checked already:

- Adapter: tried both `ember` and `zstack` before understanding that `zstack` was the supposed way to go (thanks to the `#ha-installation-addon` on Discord!)
- Adapter path: I triple checked the adapter path - both in `dmesg` and in Home Assistant _"Hardware"_ list, and I'm confident I'm using the right one - tested with both `/dev/ttyUSB0` and `/dev/serial/by-id/usb-SMLIGHT_SMLIGHT_SLZB-MR3_ea18497fa39aef119764b29061ce3355-if00-port0`

See:

```
usb 1-2: new full-speed USB device number 6 using xhci_hcd
usb 1-2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-2: Product: SMLIGHT SLZB-MR3
usb 1-2: Manufacturer: SMLIGHT
usb 1-2: SerialNumber: ea18497fa39aef119764b29061ce3355
cp210x 1-2:1.0: cp210x converter detected
usb 1-2: cp210x converter now attached to ttyUSB0
```

As per:

> When running on Linux also include output of: uname -a && cat /etc/issue.net

here it is:

```
Linux core-ssh 6.12.35-haos #1 SMP PREEMPT_DYNAMIC Wed Jul  9 12:22:07 UTC 2025 x86_64 Linux
cat: can't open '/etc/issue.net': No such file or directory
```

### Device `database.db` entry

Not found, but got this in the logs: `{"path":"/dev/ttyUSB0","manufacturer":"SMLIGHT","serialNumber":"ea18497fa39aef119764b29061ce3355","pnpId":"usb-SMLIGHT_SMLIGHT_SLZB-MR3_ea18497fa39aef119764b29061ce3355-if00-port0","vendorId":"10c4","productId":"ea60"}`

### Debug log

[z2m-debug-log.txt](https://github.com/user-attachments/files/21728448/z2m-debug-log.txt)

Horgix, 2025-08-12:
As an additional note from another experimentation, Zigbee2MQTT connects to the SLZB-MR3 fine **when using the `EFR32MG24` instead of the `CC2674P10` chip as Zigbee controller**. Might it be that the device doesn't support USB connection for the `CC2674P10` chip?

For the record, I reached out to SMLIGHT about this and will be reporting answers/findings here in followup comments.

Mrsash2020, 2025-09-21:
I am on aw 86 Win11 VMware Machine with slzb-mr3 connected via POE. I originally had the slzb-06 working fine after a few firmware updates. I moved the network to the slzb-mr3(CC2674P10 chip) and it was fine for a few hours but usually around 24 it dropped around 40 of 68 devices. 

Smlight asked me to go nuclear at this point and start fresh. So with a new pan ID etc I reset all devices and paired them again but the issue with dropped devices has not been solved. Funnily enough when I go back to slzb-06 it did not do this at all even though this time the network was being moved originally being set in mr3.

Smlight are helpful and sent me another slzb-mr3 but the issue still remains. Additional issue came up that now I cant add devices. Even the ones I remove/delete. So I am at a loss at what next since I cant nail down the  issue to z2m itself/OS/firmware as hardware is now ruled out......

One thing I noticed is the smlight integration mentions an different firmware version.

Firmware: core: v2.9.8 / zigbee: 20250212 but I am on 20240716 as per mr3 interface.

I cant even find 20250212 on the list anywhere for flashing in mr3 interface.

[image]

closed 2025-11-28 by github-actions[bot]