Koenkk/zigbee2mqtt #28251: [Wrong device]: BSEED Wall-mounted electrical socket with USB-C/A `_TZ3210_5ct6e7ye` recognize as Tuya TS011F_plug_1

MilesTEG1, 2025-08-12:
### Link

https://www.bseed.com/products/bseed-zigbee-eu-wall-sockets-type-c-with-usb-with-energy-monitoring?_pos=6&_ss=r&variant=45198394359963

### Model

GL86ZEUSKM1PDAC1W

### Description

BSEED Wall-mounted electrical EU socket with Power Monitoring 16A with USB-C/A 20W

### Vendor

BSEED

### Picture (link)

https://www.bseed.com/cdn/shop/files/1W_a1f3600f-abb9-4adc-b8a0-3ee15cb26949_1600x.jpg?v=1754288759

### Database entry

{"id":128,"type":"Router","ieeeAddr":"0xa4c138d8d9d294a3","nwkAddr":63416,"manufId":4417,"manufName":"_TZ3210_5ct6e7ye","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,1794,2820,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0018D.0\u0003","65506":54,"65508":0,"65534":0,"modelId":"TS011F","manufacturerName":"_TZ3210_5ct6e7ye","powerSource":1,"zclVersion":3,"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":""}},"haElectricalMeasurement":{"attributes":{"acCurrentDivisor":1000,"acCurrentMultiplier":1,"rmsVoltage":231,"rmsCurrent":0,"activePower":0}},"seMetering":{"attributes":{"divisor":100,"multiplier":1,"currentSummDelivered":0}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"53248":1,"53249":0,"53250":5911347,"53251":117,"53252":5559,"53253":608,"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x847127fffedfe3cd","endpointID":1},{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x847127fffedfe3cd","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x847127fffedfe3cd","endpointID":1}],"configuredReportings":[{"cluster":2820,"attrId":1285,"minRepIntval":5,"maxRepIntval":3600,"repChange":5},{"cluster":2820,"attrId":1288,"minRepIntval":5,"maxRepIntval":3600,"repChange":50},{"cluster":2820,"attrId":1291,"minRepIntval":5,"maxRepIntval":3600,"repChange":10},{"cluster":1794,"attrId":0,"minRepIntval":5,"maxRepIntval":3600,"repChange":257}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":192,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1755024663522}

### Notes

Hello,
I bought some of those wall-mounted sockets with USB-C/A 20W at AliExpress, here an example link:
https://fr.aliexpress.com/item/1005008805729264.html

It works fine I think (no power consumption measurements from USB-C/A), but has a wrong picture/vendor/model.

I hope I gave rights informations.
Thanks in advance :)

PS : here an image for the device, and the doc:

[image]

[image]

MilesTEG1, 2025-09-01:
@Koenkk Hello, Could you check this issue please?

Koenkk, 2025-09-02:
Fixed!

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `GL86ZEUSKM1PDAC1W.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2025-09-02 by Koenkk

MilesTEG1, 2025-09-02:
PR done: https://github.com/Koenkk/zigbee2mqtt.io/pull/4101