Koenkk/zigbee2mqtt #28252: Should TZ3000_femsaaua now be detected as LoraTap SC500ZB ?

ahamro, 2025-08-12:
### What happened?

My LoraTap SC500ZB switches are detected as TS130F.
Now I saw in the bug fixes of the v24.4.0 release (which I'm running): "detect: Detect _TZ3000_femsaaua as LoraTap SC500ZB"
I deleted a SC500ZB and paired it again. But it still shows as TS130F.
Is this as expected or could it be my mistake?

### What did you expect to happen?

I thought the SC500ZB would now be detected as SC500ZB, not as TS103F.

### How to reproduce it (minimal and precise)

Pair a LoraTap SC500ZB with z2m.

### Zigbee2MQTT version

2.6.0

### Adapter firmware version

20210708

### Adapter

ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

HA Blue.

Koenkk, 2025-08-13:
Yes, could you provide the data/database.db entry of this device?

ahamro, 2025-08-13:
> Yes, could you provide the data/database.db entry of this device?

Is this what you ask?

[xxx.txt](https://github.com/user-attachments/files/21761184/xxx.txt)

Koenkk, 2025-08-14:
Do you have any external converters for this device? (check if there is a `external_converters` directory in your z2m data directory).

Koenkk, 2025-08-14:
What herdsman-converters version shows up on the frontend -> settings -> about page?

ahamro, 2025-08-14:
> What herdsman-converters version shows up on the frontend -> settings -> about page?

24.11.0

Koenkk, 2025-08-16:
Its detected correctly as a SC500ZB (see the model). Note that the zigbeeModel is still TS130F (its a generic model shared by many Tuya devices), this is expected.