Koenkk/zigbee2mqtt #28267: [Feature request]: Add last_seen attribute to zigbee2mqtt/bridge/devices payload

olivaw448, 2025-08-13:
### Is your feature request related to a problem? Please describe

Currently, the zigbee2mqtt/bridge/devices topic only contains static metadata (friendly_name, ieee_address, model, etc.) and does not include the last_seen attribute, even when advanced.last_seen is enabled.

This makes it difficult for integrations (e.g., Home Assistant, Node-RED, custom dashboards) to easily detect devices that have been unavailable for a long time.
**For battery-powered devices, last_seen is crucial** to identify potential connectivity or battery issues early.

### Describe the solution you'd like

Include the last_seen attribute (using the format defined by advanced.last_seen) in each device object within the zigbee2mqtt/bridge/devices payload.
Example:
`
{
  "friendly_name": "sensor1",
  "ieee_address": "0x1234...",
  "model": "TS0201",
  ...
  "last_seen": "2025-08-14T12:34:56+02:00"
}
`

This would allow external integrations to get a complete snapshot of all devices' last activity in a **single message, without subscribing to multiple topics** or using MQTT wildcards.

### Describe alternatives you've considered

- Subscribing to zigbee2mqtt/+ to read last_seen from each device's state topic (not always possible in environments with restricted topic patterns).
- Listening to zigbee2mqtt/bridge/event to infer last_seen from device_message events, which is less accurate and requires maintaining an external cache.
- Writing a Zigbee2MQTT extension to publish a custom topic with last_seen data — possible, but duplicates functionality that could be native.
- Requesting each device's state individually — works but is inefficient for large networks; it’s more optimal to retrieve the last_seen of all devices in a single snapshot.

### Additional context

Adding last_seen to /bridge/devices will:
- Improve integration with systems like Home Assistant, Node-RED, or monitoring dashboards.
- Make it easier to automatically detect and alert about offline devices, especially battery-powered ones.
- Reduce complexity for third-party tools by avoiding the need for multiple subscriptions or custom scripts.

This would be especially useful for maintenance and reliability in Zigbee networks with many low-power sensors.

Nerivec, 2025-08-16:
`/bridge/devices` is meant to contain "static"-ish metadata. This is a potentially huge topic that should not be published too often.

olivaw448, 2025-08-16:
That makes sense. I agree it's better not to overload /bridge/devices.
However, it would still be very useful to have a way to retrieve all last_seen values in a single list from zigbee2mqtt, since this data already exists internally (and is shown in the web UI). At the moment it's not possible to get them all at once without listening to each device separately, which makes integrations and monitoring more complex.

closed 2025-11-05 by github-actions[bot]