Koenkk/zigbee2mqtt #28278: [New device support]: 1 Gang EU UK US Universal Plug Tuya ZigBee Socket with Power Monitoring and USB A and C ports

droidnexus, 2025-08-14:
### Link

https://www.aliexpress.com/item/1005007755168615.html?spm=a2g0o.order_list.order_list_main.44.4a1c1802rKCItJ

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0xa4c138404737e7a9","nwkAddr":29446,"manufId":4417,"manufName":"_TZ3000_gazjngjl","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_gazjngjl","zclVersion":3,"appVersion":67,"modelId":"TS011F","powerSource":1}},"genOnOff":{"attributes":{"32768":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002d12b331","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002d12b331","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1755212037928}

### Zigbee2MQTT version

2.6.0

### Comments

Good morning, 

I have paired this device to my Z2M. However, it's not providing any data. 

This device is showing in Zigbee2MQTT as a 2 Gang socket, which it's not. 

This is the about section of the socket in Zigbee2MQTT

Device type: Router
Zigbee Model: TS011F
Zigbee Manufacturer: _TZ3000_gazjngjl
Description: 2-gang socket with power monitoring and USB
Manufacturer : [Tuya](https://www.zigbee2mqtt.io/supported-devices/#v=Tuya)
Model: [TS011F_2_gang_power](https://www.zigbee2mqtt.io/devices/TS011F_2_gang_power.html#tuya-ts011f_2_gang_power)
Interview state: Successful

Below is the error message from the 

2025-08-14 18:53:07z2m: Publish 'get' 'power' to 'Socket' failed: 'Error: ZCL command 0xa4c138404737e7a9/1 haElectricalMeasurement.read(["activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 29446 - 1 - 167 - 2820 - 1 after 10000ms)' 2025-08-14 18:53:30z2m: Publish 'get' 'power' to 'Socket' failed: 'Error: ZCL command 0xa4c138404737e7a9/1 haElectricalMeasurement.read(["activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 29446 - 1 - 168 - 2820 - 1 after 10000ms)'

z2m: Failed to configure 'Socket', attempt 6 (Error: Device 0xa4c138404737e7a9 has no input cluster haElectricalMeasurement at getEndpointsWithCluster (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:61:15) at result.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:2048:44) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/index.ts:363:27) at processTicksAndRejections (node:internal/process/task_queues:105:5) at Configure.configure (/app/lib/extension/configure.ts:123:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:46:25) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:252:17))

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TS011F'],
    model: 'TS011F',
    vendor: '_TZ3000_gazjngjl',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2}}), m.onOff({"powerOnBehavior":false,"endpointNames":["1","2"]})],
    meta: {"multiEndpoint":true},
};
```

### What does/doesn't work with the external definition?

The power button can be turned off or on randomly in Home Assistant. The state is sometimes not reflected in HA. 

No Current / Energy / Power / Voltage information being provided 

[image]

makemelegal, 2025-08-25:
I have this device in the UK and it reports as 2 gang which it is not. Is this a Z2M  issue or manufacturer?

makemelegal, 2025-08-26:
Further to the above, I don't believe this has power monitoring but should have independent usb and plug switching through Z2M. 
the manufacturer website is here:
https://www.bseed.com/products/bseed-zigbee-uk-wall-sockets-power-outlets-kids-protection-with-pd-3-0-usb-type-c-port?srsltid=AfmBOopekfvaUueMjZeN1slCBXPnCesb3OpOW0gqC70kAXgxxcpYF9wj

Hope this helps resolve the issue and allows it to be repaired and identified correctly

closed 2025-11-02 by github-actions[bot]