Koenkk/zigbee2mqtt #28282: Losing devices since 2.6.0

fsedarkalex, 2025-08-15:
### What happened?

Hi!

sind upgrading to 2.6.0 I lost two in-wall actors, one dimmer and one switch. Both from Gledopto.

Today I was scared by an alarm that sounded like a smoke detector. After a few minutes of searching, I located the beep coming from the switch actor.

This network was running a long time without drops.

Luckily I had my logs set to debug and so I checked Z2M logs and I found something interesting (filtered):
```
[2025-08-15 15:50:49] debug: 	zh:controller:endpoint: ZCL command 0xf4b3b1fffef216b5/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-08-15 15:50:49] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xf4b3b1fffef216b5:19476/1 (0,0,2)
[2025-08-15 15:50:49] debug: 	zh:zstack:znp: --> SREQ: AF - dataRequest - {"dstaddr":19476,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":174,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,7,0,0,0]}}
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,20,76,1,1,0,0,174,0,30,5,16,7,0,0,0,208]
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2025-08-15 15:50:49] debug: 	zh:zstack:znp: <-- SRSP: AF - dataRequest - {"status":0}
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,174,104]
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,174,104]
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,174] - 104
[2025-08-15 15:50:49] debug: 	zh:zstack:znp: <-- AREQ: AF - dataConfirm - {"status":0,"endpoint":1,"transid":174}
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,20,76,0,218]
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,196,20,76,0,218]
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 196 - [20,76,0] - 218
[2025-08-15 15:50:49] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,0,0,20,76,1,1,0,76,0,160,191,197,0,0,8,24,7,1,0,0,0,32,3,20,76,29,103]
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,0,0,20,76,1,1,0,76,0,160,191,197,0,0,8,24,7,1,0,0,0,32,3,20,76,29,103]
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,0,0,20,76,1,1,0,76,0,160,191,197,0,0,8,24,7,1,0,0,0,32,3,20,76,29] - 103
[2025-08-15 15:50:49] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":0,"srcaddr":19476,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":76,"securityuse":0,"timestamp":12959648,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,7,1,0,0,0,32,3]}}
[2025-08-15 15:50:49] debug: 	zh:controller: Received payload: clusterID=0, address=19476, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=76, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":7,"commandIdentifier":1},"payload":[{"attrId":0,"status":0,"dataType":32,"attrData":3}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2025-08-15 15:50:49] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-08-15 15:50:49] debug: 	z2m: Received Zigbee message from 'Essen/Pendelleuchte', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
[2025-08-15 15:50:49] debug: 	z2m: Successfully pinged 'Essen/Pendelleuchte' (attempt 1/2)
[2025-08-15 15:50:50] debug: 	zh:zstack:unpi:parser: <-- [254,3,69,196,33,120,0,219]
[2025-08-15 15:50:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,69,196,33,120,0,219]
[2025-08-15 15:50:50] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 5 - 196 - [33,120,0] - 219
[2025-08-15 15:50:50] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd
[2025-08-15 15:50:50] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2025-08-15 15:50:50] debug: 	zh:zstack:unpi:parser: <-- [254,27,68,129,0,0,8,0,33,120,1,1,0,65,0,43,62,198,0,0,7,8,115,10,0,0,32,254,33,120,29,241]
[2025-08-15 15:50:50] debug: 	zh:zstack:unpi:parser: --- parseNext [254,27,68,129,0,0,8,0,33,120,1,1,0,65,0,43,62,198,0,0,7,8,115,10,0,0,32,254,33,120,29,241]
[2025-08-15 15:50:50] debug: 	zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,8,0,33,120,1,1,0,65,0,43,62,198,0,0,7,8,115,10,0,0,32,254,33,120,29] - 241
[2025-08-15 15:50:50] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":8,"srcaddr":30753,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":65,"securityuse":0,"timestamp":12992043,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,115,10,0,0,32,254]}}
[2025-08-15 15:50:50] debug: 	zh:controller: Received payload: clusterID=8, address=30753, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=65, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":115,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":32,"attrData":254}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2025-08-15 15:50:50] debug: 	zh:controller:endpoint: ZCL command 0x4c5bb3fffec28078/1 genLevelCtrl.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"reservedBits":0,"transactionSequenceNumber":115,"writeUndiv":false})

```

a little later

```
[2025-08-15 15:51:13] debug: 	zh:zstack:unpi:parser: --- parseNext [254,33,68,129,0,0,0,239,113,150,1,1,0,65,0,150,26,218,0,0,13,9,20,2,0,134,9,2,0,4,0,0,1,55,113,150,29,172,254,13,69,201,20,76,181,22,242,254,255,177,179,244,0,0,0,127,254,3,69,196,233,59,0,80]
[2025-08-15 15:51:13] debug: 	zh:zstack:unpi:parser: --> parsed 33 - 2 - 4 - 129 - [0,0,0,239,113,150,1,1,0,65,0,150,26,218,0,0,13,9,20,2,0,134,9,2,0,4,0,0,1,55,113,150,29] - 172
[2025-08-15 15:51:13] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":61184,"srcaddr":38513,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":65,"securityuse":0,"timestamp":14293654,"transseqnumber":0,"len":13,"data":{"type":"Buffer","data":[9,20,2,0,134,9,2,0,4,0,0,1,55]}}
[2025-08-15 15:51:13] debug: 	zh:controller: Received payload: clusterID=61184, address=38513, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=65, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":20,"commandIdentifier":2},"payload":{"seq":34304,"dpValues":[{"dp":9,"datatype":2,"data":{"type":"Buffer","data":[0,0,1,55]}}]},"command":{"ID":2,"parameters":[{"name":"seq","type":33},{"name":"dpValues","type":1011}],"name":"dataReport"}}
[2025-08-15 15:51:13] debug: 	zh:zstack:unpi:parser: --- parseNext [254,13,69,201,20,76,181,22,242,254,255,177,179,244,0,0,0,127,254,3,69,196,233,59,0,80]
[2025-08-15 15:51:13] debug: 	zh:zstack:unpi:parser: --> parsed 13 - 2 - 5 - 201 - [20,76,181,22,242,254,255,177,179,244,0,0,0] - 127
[2025-08-15 15:51:13] debug: 	zh:zstack:znp: <-- AREQ: ZDO - leaveInd
```

Then, the following is repeating multiple times:
```
[2025-08-15 15:51:13] debug: 	zh:controller: Device leave '0xf4b3b1fffef216b5'
[2025-08-15 15:51:13] debug: 	zh:controller: Device leave is from unknown or already deleted device '0xf4b3b1fffef216b5'
[... cut ...] 9,244,0,0,0,127,254,3,69,196,233,59,0,80,254,5,69,196,211,66,1,167,119,196,254,5,69,196,211,66,1,167,119,196,254,3,69,196,233,59,0,80,254,5,69,196,140,45,1,60,245,237]
[... 30 lines cut ...]
I don't know... Happens sporadic

### Zigbee2MQTT version

2.6.0

### Adapter firmware version

20240710

### Adapter

SONOFF using ZStack3x0

### Setup

HA AddOn on debian on proxmox. Coordinator using ser2net - no virtual USB crap

### Device `database.db` entry

lost :)

### Debug log

see "what happened"

AlexMKX, 2025-08-17:
Same here.
Different devices from different manufacturers are unable to join. They joining and leaving network afterwards. Doing this all the time till join is available (e.g. 5 mins)

`warning 2025-08-17 18:35:05z2m: Device '0xa4c1385195d84a47' left the network
info 2025-08-17 18:35:05z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385195d84a47","ieee_address":"0xa4c1385195d84a47"},"type":"device_leave"}'
info 2025-08-17 18:35:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385195d84a47', payload '{"current":null,"energy":null,"last_seen":"2025-08-17T18:35:07+03:00","linkquality":47,"power":null,"power_outage_memory":null,"state":"OFF","switch_type":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":221}'
info 2025-08-17 18:35:09zh:controller: Interview for '0xa4c1385195d84a47' started
info 2025-08-17 18:35:09z2m: Device '0xa4c1385195d84a47' joined
info 2025-08-17 18:35:09z2m: Starting interview of '0xa4c1385195d84a47'
info 2025-08-17 18:35:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385195d84a47","ieee_address":"0xa4c1385195d84a47"},"type":"device_joined"}'
info 2025-08-17 18:35:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385195d84a47","ieee_address":"0xa4c1385195d84a47","status":"started"},"type":"device_interview"}'
info 2025-08-17 18:35:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385195d84a47', payload '{"current":null,"energy":null,"last_seen":"2025-08-17T18:35:08+03:00","linkquality":47,"power":null,"power_outage_memory":null,"state":"OFF","switch_type":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":221}'
info 2025-08-17 18:35:09z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385195d84a47","ieee_address":"0xa4c1385195d84a47"},"type":"device_announce"}'
info 2025-08-17 18:35:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385195d84a47', payload '{"current":null,"energy":null,"last_seen":"2025-08-

17T18:35:08+03:00","linkquality":51,"power":null,"power_outage_memory":null,"state":"OFF","switch_type":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":221}'
info 2025-08-17 18:35:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385195d84a47', payload '{"current":null,"energy":null,"last_seen":"2025-08-17T18:35:08+03:00","linkquality":47,"power":null,"power_outage_memory":null,"state":"OFF","switch_type":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":221}'
info 2025-08-17 18:35:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c1385195d84a47', payload '{"current":null,"energy":null,"last_seen":"2025-08-17T18:35:08+03:00","linkquality":54,"power":null,"power_outage_memory":null,"state":"OFF","switch_type":null,"update":{"installed_version":-1,"latest_version":-1,"state":null},"voltage":221}'
info 2025-08-17 18:35:10zh:controller: Succesfully interviewed '0xa4c1385195d84a47'
info 2025-08-17 18:35:10z2m: Successfully interviewed '0xa4c1385195d84a47', device has successfully been paired
info 2025-08-17 18:35:10z2m: Device '0xa4c1385195d84a47' is supported, identified as: Tuya Switch with power monitoring (TS0001_power)
info 2025-08-17 18:35:10z2m: Configuring '0xa4c1385195d84a47'
info 2025-08-17 18:35:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Switch with power monitoring","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","label":"State","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":1,"description":"Instantaneous measured power","label":"Power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Instantaneous measured electrical current","label":"Current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Measured electrical potential value","label":"Voltage","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","label":"Energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"access":7,"description":"Type of the switch","label":"Switch type","name":"switch_type","property":"switch_type","type":"enum","values":["toggle","state","momentary"]},{"access":7,"description":"Recover state after power outage","label":"Power outage memory","name":"power_outage_memory","property":"power_outage_memory","type":"enum","values":["on","off","restore"]},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0001_power","options":[{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","label":"Power calibration","name":"power_calibration","property":"power_calibration","type":"numeric","value_step":0.1},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device. This option can only decrease the precision, not increase it.","label":"Power precision","name":"power_precision","property":"power_precision","type":"numeric","value_max": [... cut ...] stalled_version":-1,"latest_version":-1,"state":null},"voltage":222}'
warning 2025-08-17 18:35:30z2m: Device '0xa4c1385195d84a47' left the network
info 2025-08-17 18:35:30z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1385195d84a47","ieee_address":"0xa4c1385195d84a47"},"type":"device_leave"}'`

[45df7312_zigbee2mqtt_2025-08-17T15-35-53.907Z.log](https://github.com/user-attachments/files/21823246/45df7312_zigbee2mqtt_2025-08-17T15-35-53.907Z.log)

asmolarczyk, 2025-08-19:
Wow! This is really interesting. How did found the cause of the issue?!
Unfortunately my problems seem to be related to this...

Today I had a power breakdown. When power came back, I had the same problem with lost devices as a few days ago. In the meanwhile I enabled debug logging and imho this is interesting:

```
[2025-08-19 18:53:59] debug: 	zh:controller: Received ZDO response: clusterId=END_DEVICE_ANNOUNCE, status=SUCCESS, payload={"nwkAddress":55380,"eui64":"0x001788010b7fc987","capabilities":{"alternatePANCoordinator":0,"deviceType":1,"powerSource":1,"rxOnWhenIdle":1,"reserved1":0,"reserved2":0,"securityCapability":0,"allocateAddress":1}}
[2025-08-19 18:53:59] debug: 	zh:controller: Device announce from '0x001788010b7fc987:55380'
[2025-08-19 18:53:59] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer/Light/Esstisch_1', payload '{"brightness":254,"color":{"h":32,"hue":32,"s":81,"saturation":81,"x":0.4574,"y":0.41},"color_mode":"color_temp","color_temp":366,"effect":null,"hue_power_on_behavior":"off","last_seen":"2025-08-19T16:53:59.237Z","linkquality":25,"power_on_behavior":"off","state":"OFF","update":{"installed_version":16787202,"latest_version":16787202,"state":"idle"}}'
[2025-08-19 18:53:59] debug: 	z2m: Device 'Wohnzimmer/Light/Esstisch_1' announced itself
```
and a little bit later:

```
[2025-08-19 18:54:00] debug: 	zh:controller: Received ZDO response: clusterId=END_DEVICE_ANNOUNCE, status=SUCCESS, payload={"nwkAddress":10191,"eui64":"0xd85def11a1001668","capabilities":{"alternatePANCoordinator":0,"deviceType":1,"powerSource":1,"rxOnWhenIdle":1,"reserved1":0,"reserved2":0,"securityCapability":0,"allocateAddress":1}}
[2025-08-19 18:54:00] debug: 	zh:controller: Device announce from '0xd85def11a1001668:10191'
[2025-08-19 18:54:00] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Wohnzimmer/Switch/Schalter', payload '{"last_seen":"2025-08-19T16:54:00.208Z","linkquality":69}'
[2025-08-19 18:54:00] debug: 	zh:controller: Device '0xd85def11a1001668' got new networkAddress '10191'
[2025-08-19 18:54:00] debug: 	zh:controller:database: Writing database to '/app/data/database.db'
[2025-08-19 18:54:00] debug: 	z2m: Device 'Wohnzimmer/Switch/Schalter' changed network address
[2025-08-19 18:54:00] debug: 	z2m: Device 'Wohnzimmer/Switch/Schalter' announced itself
```

Wohnzimmer/Light/Esstisch_1 is one of the philips bulbs not connected to the network anymore and Wohnzimmer/Switch/Schalter is a Busch-Jaeger Wall-Switch acting as a router for Wohnzimmer/Light/Esstisch_1.
Why did the Wall-Switch got a new network address? Is this the cause for the 'MAC no ack' messages?
And why are scenes/groups of the bulb lost after repairing?

fsedarkalex, 2025-08-19:
My coordinator backup contains 115 entries while the network consists of 84 devices currently.
So I am far from the 200 in both nubers.

For cleaning: Is there some way to automatically clean it up or is it a manual task?

AlexMKX, 2025-08-29:
@thogens  it depends on firmware and stick memory . In generally with current koenkk firmware settings, as far as I can understand you're near the limit. 
I'll try SLZB it looks it could handle bigger ZB networks (like 400-500 devices) and yet the routing will be more efficient. Again, as far as I can understand the more devices you have the bigger routing table there is. If coordinator memory is not enough, some routes are discarded from time to time. As a result packets instead of going directly from source to target can go broadcarst-way and flood the network slowing things down.

thogens, 2025-09-10:
Update: After migrating 130 devices to new stick, I could perfectly pair devices to the old stick again.
I did not change any other settings. - Only deleting devices to migrate to new one.

fsedarkalex, 2025-11-23:
I re-paired the currently affected device. It is this type: https://www.zigbee2mqtt.io/devices/GL-SD-001.html

So we have two different vendors but similar device type.
Maybe they are sharing something common?

I also had plug-in smart sockets from mmultiple verndors affected but I phased them out as they had other issues, too. But maybe related.

closed 2026-01-30 by github-actions[bot]