Koenkk/zigbee2mqtt #28289: TS0601_3_phase_clamp_meter Wrong power readings, is not working

carlosduque-incoxe, 2025-08-16:
### What happened?

Hello! I have three [TS0601 3-phase clamp meters](https://www.zigbee2mqtt.io/devices/TS0601_3_phase_clamp_meter.html#tuya-ts0601_3_phase_clamp_meter):

One rated at 250A

One rated at 500A

And one rated at 700A

I use them across several installations that reach those consumption levels.

The WiFi version supports bidirectional measurement (for example, in a solar installation), but the Zigbee version, which I’m using, does not.

When connected to a Tuya hub via Zigbee, the devices display correct values in the Tuya app.
However, when connected to Zigbee2MQTT, the readings make no sense—showing very different power values between phases, randomly fluctuating current measurements, incorrect voltages, and so on.

It’s important to note that the device is powered by 3 phases + neutral, and here in Spain we have 400V between phases and 230V between phase and neutral, which must also be taken into account when calculating power.

I have the Tuya API DPs available along with the reported values, in case they might be useful.
I’m attaching them here. `{
  "result": {
    "properties": [
      {
        "code": "forward_energy_total",
        "custom_name": "",
        "dp_id": 1,
        "time": 1753367257011,
        "type": "value",
        "value": 0
      },
      {
        "code": "month_energy",
        "custom_name": "",
        "dp_id": 3,
        "time": 1753367248641,
        "type": "raw"
      },
      {
        "code": "daily_energy",
        "custom_name": "",
        "dp_id": 4,
        "time": 1753367248641,
        "type": "raw"
      },
      {
        "code": "phase_a",
        "custom_name": "",
        "dp_id": 6,
        "time": 1753367473927,
        "type": "raw",
        "value": "CSQAAAAAAAA="
      },
      {
        "code": "phase_b",
        "custom_name": "",
        "dp_id": 7,
        "time": 1753367476989,
        "type": "raw",
        "value": "CSQAAAAAAAA="
      },
      {
        "code": "phase_c",
        "custom_name": "",
        "dp_id": 8,
        "time": 1753367480047,
        "type": "raw",
        "value": "CSMAAAAAAAA="
      },
      {
        "code": "power_total",
        "custom_name": "",
        "dp_id": 9,
        "time": 1753367253978,
        "type": "value",
        "value": 0
      },
      {
        "code": "EnergyConsumedA",
        "custom_name": "",
        "dp_id": 101,
        "time": 1753367260060,
        "type": "value",
        "value": 3
      },
      {
        "code": "PowerFactorA",
        "custom_name": "",
        "dp_id": 102,
        "time": 1753367269210,
        "type": "value",
        "value": 0
      },
      {
        "code": "EnergyConsumedB",
        "custom_name": "",
        "dp_id": 111,
        "time": 1753367263102,
        "type": "value",
        "value": 3
      },
      {
        "code": "PowerFactorB",
        "custom_name": "",
        "dp_id": 112,
        "time": 1753367272274,
        "type": "value",
        "value": 0
      },
      {
        "code": "EnergyConsumedC",
        "custom_name": "",
        "dp_id": 121,
        "time": 1753367266154,
        "type": "value",
        "value": 3
      },
      {
        "code": "PowerFactorC",
        "custom_name": "",
        "dp_id": 122,
        "time": 1753367248641,
        "type": "value",
        "value": 1
      },
      {
        "code": "Current",
        "custom_name": "",
        "dp_id": 131,
        "time": 1753367324219,
        "type": "value",
        "value": 0
      },
      {
        "code": "Frequency",
        "custom_name": "",
        "dp_id": 132,
        "time": 1753367281439,
        "type": "value",
        "value": 50
      },
      {
        "code": "Temperature",
        "custom_name": "",
        "dp_id": 133,
        "time": 1753367513668,
        "type": "value",
        "value": 239
      },
      {
        "code": "DeviceStatus",
        "custom_name": "",
        "dp_id": 134,
        "time": 1753367287554,
        "type": "bitmap",
        "value": 0
      }
    ]
  },
`

[image]

[image]

### Zigbee2MQTT version

2.6.0

### Adapter firmware version

20240710

### Adapter

ZStack3x0

### Setup

It is on home assistant os addon in a pc x86  32gb of ram

vally2911, 2025-09-05:
I have the same Device with the same Problem

mwdmwd, 2025-11-02:
These per-phase DP values in the API dump are all zero except the voltage (so power and current are zero),

```
>>> buf = base64.b64decode("CSQAAAAAAAA=")
>>> (buf[1] | (buf[0] << 8)) / 10
234.0
>>> buf[2:]
b'\x00\x00\x00\x00\x00\x00'
```

@carlosduque-incoxe, could you provide a similar raw dump (or a few) while there is a sigificant load, ideally along with the correct measured values as displayed in the Tuya app?

On my end I see reasonable values displayed for all parameters, but I have only used loads of around 10 amps at most, maybe the measurement values have more discontinuities than those handled as part of #18603?

See also #26694

closed 2026-02-07 by github-actions[bot]