Koenkk/zigbee2mqtt #28293: Dev branch tip reporting many "out of range" errors that prevent events from being emitted

rccoleman, 2025-08-16:
### What happened?

With the latest dev branch container, I'm seeing quite a lot of this type of error in the log, and my motion sensors and remotes have become non-functional or only partially functional.

```
[2025-08-16 08:55:50] error:    z2m: Publish 'set' 'state' to 'Kit Light Group' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 6 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":655350}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received 655350)'
[2025-08-16 08:55:54] error:    z2m: Publish 'set' 'state' to 'Kit Light Group' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 6 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":655350}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received 655350)'
[2025-08-16 08:55:55] error:    z2m: Publish 'set' 'state' to 'Kit Light Group' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 6 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":655350}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received 655350)'
[2025-08-16 08:55:56] error:    z2m: Publish 'set' 'state' to 'Kit Light Group' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 6 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":655350}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received 655350)'
[2025-08-16 09:20:04] error:    z2m: Publish 'set' 'state' to 'DNH Group' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 8 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":655350}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received 655350)'
[2025-08-16 09:20:32] error:    z2m: Publish 'set' 'state' to 'Kit Light Group' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 6 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":655350}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received 655350)'
[2025-08-16 09:20:37] error:    z2m: Publish 'set' 'state' to 'DNH Group' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 8 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":655350}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received 655350)'
[2025-08-16 09:24:00] error:    z2m: Publish 'set' 'state' to 'Kit Light Group' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 6 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":655350}) failed (The value of "value" is out of range. It must be >= 0 and <= 65535. Received 655350)'
```

See full log below with debug enabled.  It looks like maybe something is now multiplying the actual value by 10 by mistake.

My Aqara motion P1 sensors (lumi.motion.ac02) report that error when motion is detected and don't emit an event, and on a Tuya 3-button remote, only button #2 works, but 1 and 3 result in this "out of range" error.

The problem goes away if I revert back to the release container (

### What did you expect to happen?

I expected all my devices to continue to behave properly, as they do in the release version of Z2M.

### How to reproduce it (minimal and precise)

Trigger motion with an Aqara P1 PIR motion sensor and observe the "out of range" error shows up in the log and the state change is not published.

### Zigbee2MQTT version

2.6.0-dev commit: 4c9784255bab79d7d8a394135bfd6f5b9e40ef9b

### Adapter firmware version

20240710 (zStack)

### Adapter

Tubeszb Ethernet cc2652p7

### Setup

Zigbee2MQTT on a Debian-based mini-PC

Linux rcoleman-linux 6.1.0-20-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.85-1 (2024-04-11) x86_64 GNU/Linux
Debian GNU/Linux 12

### Debug log

[log.log](https://github.com/user-attachments/files/21815051/log.log)

rccoleman, 2025-08-19:
Ah, that looks promising!  I have Inovelli Blue switches throughout my house, and all cases of this are controlling them.  Will check out the dev branch later today.

rccoleman, 2025-08-20:
The latest dev container is working fine for me, so I'm sure that was it. Thanks!

rccoleman, 2025-08-20:
While that change fixes that specific error, I'm still having significant issues with my Inovelli switches that I don't see with the released version. I need to try to narrow it down, but my 'night' routine that turns on several groups with mostly Inovelli switches didn't turn on most of the lights tonight. I reverted back to the release version and the same routine worked fine.

Koenkk, 2025-08-20:
Could you provide the debug log of both the latest release and the dev branch when doing this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

@rohankapoorcom  I think this is some regression from https://github.com/Koenkk/zigbee-herdsman-converters/pull/9757

rccoleman, 2025-08-20:
I was planning on opening a new issue, as this specific one is fixed. I really just noticed the breakage before I went to bed last night and haven't had any time to investigate further. I will do so later tonight.

InovelliUSA, 2025-08-21:
I believe that the problem is we are sending "move to level with on/off" with a level of 0 to turn the switch off and not using the On Off cluster. The advantage was that bound devices get sent the transition speed when turning off. The unintended consequence is that I think the switch is retaining the 0% level when it turns back on. @rohankapoorcom we should return the converter to using just on / off to see if that fixes the problem. I think I discussed this with you the other day before we knew of the unintended consequence.

rohankapoorcom, 2025-08-21:
I will defer to @Koenkk to understand how to override this behavior based on how the default converter operates. Do we need to add the on_off converter to the devices as well [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/4a2a5480c98abfe534098c61b6bae7c56a328728/src/devices/inovelli.ts#L2530)? Or is there a better way to handle this?

rccoleman, 2025-08-21:
Sure, I'll try it tonight.

rccoleman, 2025-08-22:
Seems to help with both the on-level and transition time.  I'll keep testing around the house.

rccoleman, 2025-08-22:
I've been running the dev branch full time since yesterday and haven't seen any incorrect behavior, but I also don't use dimming or transition times. For my use cases this small change is working fine, and I can try to pull down the PR and test if it's in good shape.

rccoleman, 2025-08-30:
Resolved via https://github.com/Koenkk/zigbee-herdsman-converters/pull/9835

closed 2025-08-30 by rccoleman