Koenkk/zigbee2mqtt #28297: zh:ember: Failed to register group in multicast table with status=INVALID_STATE.

Nopel2020, 2025-08-17:
### What happened?

I am running 2.6.0. I have several zigbee groups defined. Whenever I toggle a group, I get a warning like this in the log:
```
[2025-08-17 00:15:26] error: 	zh:ember: Failed to register group '14' in multicast table with status=INVALID_STATE.
[2025-08-17 00:31:38] error: 	zh:ember: Failed to register group '11' in multicast table with status=INVALID_STATE.
[2025-08-17 05:59:18] error: 	zh:ember: Failed to register group '14' in multicast table with status=INVALID_STATE.
[2025-08-17 06:17:51] error: 	zh:ember: Failed to register group '22' in multicast table with status=INVALID_STATE.
[2025-08-17 06:17:51] error: 	zh:ember: Failed to register group '22' in multicast table with status=INVALID_STATE.
[2025-08-17 06:49:00] error: 	zh:ember: Failed to register group '22' in multicast table with status=INVALID_STATE.
```
The groups seem to function correctly however. Each groups contains either several identical smart switches (Sonoff ZBMINIR2) or several identical bulbs (Philips Hue). I'm using an SLZB-06M coordinator running the latest firmware (20250220 and v2.9.8).

I see someone else already reported this issue for 2.3.0, but there was no response and the issue was closed as stale (https://github.com/Koenkk/zigbee2mqtt/issues/27339). It is still an issue for me today on 2.6.0.

### What did you expect to happen?

The groups seem to function fine, so I'd expect the error log to stay empty.

### Zigbee2MQTT version

2.6.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

EmberZNet

### Setup

I'm running zigbee2mqtt as an add-on on Home Assistant OS.

Nopel2020, 2025-08-22:
Anyone have the same problem? I've deleted the groups and recreated them with different IDs, but still the logs are completely filling up with these errors:

```
[2025-08-21 20:02:52] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:02:52] error: 	zh:ember: Failed to register group '12' in multicast table with status=INVALID_STATE.
[2025-08-21 20:02:52] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:02] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:02] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:02] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:03] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:10] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:10] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:11] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:11] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:19] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:19] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:19] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:20] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:23] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:23] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:24] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:24] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:25] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:25] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:26] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:28] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:29] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:29] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:29] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:32] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:32] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:33] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:33] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:35] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:35] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:36] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:38] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:38] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:38] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:39] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:41] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:41] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:42] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:42] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:45] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:45] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:46] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:49] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 20:03:49] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[... cut ...]
[2025-08-21 23:32:03] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 23:40:02] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-21 23:40:02] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-22 06:37:38] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
[2025-08-22 06:37:39] error: 	zh:ember: Failed to register group '23' in multicast table with status=INVALID_STATE.
```

Vospe1, 2025-08-24:
I have exactly the same issue, but with the Sonoff Dongle E. Latest firmware from Sonoff (8.0.2). 
Thinking of trying a different firmware (either 7.4.4, or one from https://github.com/darkxst/), but worried I will loose my devices)

Diaoul, 2025-09-02:
I had this issues as well. Removed the faulty device, rebooted Zigbee2MQTT and repaired. Problem gone.

Nopel2020, 2025-09-02:
You misunderstand the issue. There is no faulty device. This is happening to all of my zigbee groups, not to any of my devices.

Nerivec, 2025-09-30:
The error is straight from the coordinator (this is not happening on Z2M side). It is a generic one though, so hard to say what's going on exactly. I'll check the source code, see if we can narrow  down where this might be happening, and why.

Have you installed the latest 8.0.2 on the coordinator just in case?
https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update7

You can also try the pre-release, lots of bugfixes from silabs since (with the caveat mentioned in the release note):
https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2025.6.1-pre0

Nopel2020, 2025-10-01:
Thanks for looking into this! I'm running the latest available OTA coordinator firmware:
Revision: 20250220
SDK 8.0.2

Diaoul, 2025-10-01:
@Nerivec what do your firmware versions bring vs official ones? More recent bleeding edge builds? Focus on stability/performance/something else?

Nopel2020, 2025-10-01:
> [@Nopel2020](https://github.com/Nopel2020) have you tried removing all the devices from the group, remove the group then recreate everything? I know it's painful but it did work for me 🤷

I have, several times! That's why the ids of the groups are into the twenties by now 😭

Nopel2020, 2025-10-01:
@Nerivec I wasn't aware of your builds. I just flashed your v2025.6.1-pre0 an hour ago. So far nothing in the logs, but if it happens again, I'll update and send a tokens backup.

Nopel2020, 2025-10-04:
Well, it's been 3 days, and so for no errors/warnings in the logs... Looking good so far with v2025.6.1-pre0 on the SLZB-06M.