Koenkk/zigbee2mqtt #28298: [New device support]: Hue Play wall washer LGT009

0rsa, 2025-08-17:
### Link

https://www.philips-hue.com/fr-fr/p/hue-white-and-color-ambiance-projecteur-mural-hue-play/8720169385993

### Database entry

{"id":93,"type":"Router","ieeeAddr":"0x001788010e541b4c","nwkAddr":6969,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LGT009","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513,64516],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"LGT009","manufacturerName":"Signify Netherlands B.V.","powerSource":1,"zclVersion":8,"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20240902","swBuildId":"1.122.2"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":0,"dateCode":"20240902","swBuildId":"1.122.2","zclVersion":8,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1755413678549}

### Zigbee2MQTT version

2.6.0

### Comments

This is a special lamp because it has 3 independent lights built in. However, I only see one lamp and I can only change the color of all the lamps at the same time.

There may be some special work to be done on this model.

If you need more information, don't hesitate. Thank you.

### External definition

```shell
export default {
    zigbeeModel: ['LGT009'],
    model: 'LGT009',
    vendor: 'Signify Netherlands B.V.',
    description: 'Automatically generated definition',
    extend: [philips.m.light({"colorTemp":{"range":[153,500]},"color":{"modes":["xy","hs"],"enhancedHue":true}})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

I can turn off or change the color of the lamp but I cannot control the built-in lights individually

decowboy, 2025-10-04:
I'm also using this device and am seeing the same limitation.

My model however is reported not as **LGT009**, but as **LGT010**, and I see it in zigbee-herdsman-converters as model [**LGT012**](https://github.com/Koenkk/zigbee-herdsman-converters/pull/9698). As such the device is currently not recognized in Zigbee2MQTT, even though it is listed as supported.

Koenkk, 2025-10-04:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/philips.ts)?

topaDev, 2025-11-02:
just wondering if it is now possible to control all 3 LED independently, as there now seems to be support with version 2.6.3?
As stated in https://github.com/Koenkk/zigbee2mqtt/issues/29107