Koenkk/zigbee2mqtt #28319: [New device support]: External converter that adds support for Nova Digital's tubular blind motor.

weberamaral, 2025-08-19:
### Link

https://www.novadigitalsmart.com.br/produtos/motor-tubular-inteligente-a-bateria-zbcmr-02

### Database entry

{"id":16,"type":"EndDevice","ieeeAddr":"0xa4c13845898acfd6","nwkAddr":15267,"manufId":4742,"manufName":"_TZE600_5cu64znk","powerSource":"Battery","modelId":"TS0105","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"genPollCtrl":{"attributes":{"checkinInterval":14400}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0037e74674","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":79,"stackVersion":2,"hwVersion":0,"zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1755514993194,"checkinInterval":3600}

### Zigbee2MQTT version

2.6.0

### Comments

Hi, 

First of all, please excuse my English — I needed Google Translate a lot.

Within the support of numerous Tuya devices, the previous model is already listed on the Zigbee2MQTT website [https://www.zigbee2mqtt.io/devices/ZBCMR-01.html](https://www.zigbee2mqtt.io/devices/ZBCMR-01.html) .However, this is a new battery-powered version, and when I tried to pair it, it identified it as a generic switch device.

Therefore, I created an external converter to correctly support this new version, in addition to displaying the battery status.

If everything is correct, I would need help contributing this support to the new version.

How can I proceed?

### External definition

```shell
import {fromZigbee, toZigbee} from "zigbee-herdsman-converters/lib/legacy";
import {access as ea, presets as e} from "zigbee-herdsman-converters/lib/exposes";
import * as tuya from "zigbee-herdsman-converters/lib/tuya";

export default {
    fingerprint: tuya.fingerprint("TS0105", ["_TZE600_5cu64znk"]),
    model: "ZBCMR-02",
    vendor: "Nova Digital",
    description: "Motor Tubular Inteligente à Bateria Zigbee",
    fromZigbee: [fromZigbee.tuya_cover, tuya.fz.datapoints],
    toZigbee: [toZigbee.tuya_cover_control, tuya.tz.datapoints],
    exposes: [e.cover_position().setAccess("position", ea.STATE_SET), e.battery()],
    meta: {
        tuyaDatapoints: [[13, "battery", tuya.valueConverter.raw]],
    },
};
```

### What does/doesn't work with the external definition?

Adds support for Nova Digital's Tuya smart tubular motor. This latest version is battery-powered.

In addition to enabling support, it also displays the battery status.

This allows you to "open," "close," "stop," and set the position.

eduseiden, 2025-08-19:
Hi @weberamaral 

Try this external converter. Just adjust the fingerprint. I use this code for Novadigital blinds, but I commented the battery lines. Here, I post the original code for you to test.

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definitions = [
  {
    fingerprint: [
        {modelID: 'TS0601', manufacturerName: '_TZE200_9vpe3fl1'},
    ],
    whiteLabel: [
        tuya.whitelabel('Zemismart', 'ZM25RX-08/30', 'Tubular motor', ['_TZE200_bv1jcqqu', '_TZE200_7eue9vhc']),
    ],
    model: 'TS0601_cover_8',
    vendor: 'TuYa',
    description: 'Cover motor',
    onEvent: tuya.onEvent(),
    configure: tuya.configureMagicPacket,
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    options: [exposes.options.invert_cover()],
    exposes: [
        //e.text('work_state', ea.STATE),
        e.cover_position().setAccess('position', ea.STATE_SET),
        e.battery(),
        // it's also has `UPPER`/`LOWER` command that can be used to move motor over set limit, but it's not clear why :)
        e.enum('program', ea.SET, ['SET BOTTOM', 'SET UPPER', 'RESET', /* 'UPPER', 'LOWER', */]).withDescription('Set the upper/bottom limit'),
        e.enum('motor_direction', ea.STATE_SET, ['NORMAL', 'REVERSED']).withDescription('Motor direction'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverterBasic.lookup({'OPEN': tuya.enum(0), 'STOP': tuya.enum(1), 'CLOSE': tuya.enum(2)})],
            [2, 'position', tuya.valueConverter.coverPosition],
            [3, 'position', tuya.valueConverter.coverPosition],
            [5, 'motor_direction', tuya.valueConverterBasic.lookup({'NORMAL': tuya.enum(0), 'REVERSED': tuya.enum(1)})],
            [7, 'work_state', tuya.valueConverterBasic.lookup({'standby': tuya.enum(0), 'success': tuya.enum(1), 'learning': tuya.enum(2)})],
            [13, 'battery', tuya.valueConverter.raw],
            [101, 'program', tuya.valueConverterBasic.lookup({'SET BOTTOM': tuya.enum(0), 'SET UPPER': tuya.enum(1), 'RESET': tuya.enum(4), 'LOWER': tuya.enum(2), 'UPPER': tuya.enum(3)})],
        ],
    },
  },
]

module.exports = definitions;`

weberamaral, 2025-08-19:
Hi,

okay, I'll make the adjustments and test it.

weberamaral, 2025-09-05:
Hi, @eduseiden 

This external converter was working perfectly before version 2.6.1, but after the update it stopped working.
I noticed that the external converter was disabled and the device was recognized as a 4-key switch.

Is there any way to recover it without reverting the update?