Koenkk/zigbee2mqtt #28331: No electric measurement from sunricher switch SR-ZG9101SAC-HP-SWITCH-2CH

Choun8, 2025-08-20:
### What happened?

Hello guys,
I bought 2 switches SR-ZG9101SAC-HP-SWITCH-2CH and I don't get any electric measurement.
The detection and addition of the devices went well, they work (on/off).
Unfortunately, I get an error about the measurement config (I don't know how to explain correctly as I am a newbie in Z2M, sorry) and current, voltage etc are "Null" in the "expose" tab.

Could you help ?

Here is what I get for the "generate external definition" :
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['ON/OFF(2CH)'],
    model: 'ON/OFF(2CH)',
    vendor: 'Sunricher',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2,"11":11}}), m.onOff({"powerOnBehavior":false,"endpointNames":["1","2"]}), m.electricityMeter()],
    meta: {"multiEndpoint":true},
};

### What did you expect to happen?

Get the electrical measurement values

### Zigbee2MQTT version

2.6.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

Ember, on a Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2 ZBDONGLE-E 

### Setup

I am running Z2M as an integration in Home Assistant.
Home assistant is running on the a "Home Assistant Green" box.

### Device `database.db` entry

{"id":12,"type":"Router","ieeeAddr":"0xb43a31fffece7290","nwkAddr":4959,"manufId":4644,"manufName":"Sunricher","powerSource":"Mains (single phase)","modelId":"ON/OFF(2CH)","epList":[1,2,11,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"modelId":"ON/OFF(2CH)","manufacturerName":"Sunricher","powerSource":1,"zclVersion":3,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r5"}},"genOnOff":{"attributes":{"onOff":0,"startUpOnOff":255}},"haElectricalMeasurement":{"attributes":{}}},"binds":[{"cluster":2820,"type":"endpoint","deviceIeeeAddress":"0x08b95ffffe596cd2","endpointID":1},{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x08b95ffffe596cd2","endpointID":1},{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x08b95ffffe596cd2","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"modelId":"ON/OFF(2CH)","manufacturerName":"Sunricher","powerSource":1,"zclVersion":3,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r5"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x08b95ffffe596cd2","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"11":{"profId":260,"epId":11,"devId":7,"inClusterList":[0,1794,2820,2821,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"modelId":"ON/OFF(2CH)","manufacturerName":"Sunricher","powerSource":1,"zclVersion":3,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r5"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":1,"dateCode":"NULL","swBuildId":"2.9.2_r5","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1755669900398}

### Debug log

error 2025-08-20 08:54:15z2m: Failed to configure 'Commutateur Vortice atelier', attempt 1 (Error: ZCL command 0xb43a31fffece7290/1 haElectricalMeasurement.read(["acVoltageMultiplier","acVoltageDivisor","acCurrentMultiplier"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:372:28) at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1005:26) at processTicksAndRejections (node:internal/process/task_queues:105:5) at runNextTicks (node:internal/process/task_queues:69:3) at processImmediate (node:internal/timers:453:9) at Endpoint.read (/app/node_modules/.pnpm/zigbee-herdsman@5.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:481:29) at Object.readEletricalMeasurementMultiplierDivisors (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:26:5) at Object.configure (/app/node_modules/.pnpm/zigbee-herdsman-converters@24.11.0/node_modules/zigbee-herdsman-converters/src/devices/sunricher.ts:1314:13) at Configure.configure (/app/lib/extension/configure.ts:123:13) at Immediate.<anonymous> (/app/lib/extension/configure.ts:66:17))
info 2025-08-20 08:55:42z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Mur SAM', payload '{"ballast_maximum_level":null,"ballast_minimum_level":null,"ballast_status_lamp_failure":false,"ballast_status_non_operational":false,"brightness":115,"level_config":

chris-1243, 2025-08-28:
Could you add this code as an external converter with a `*.mjs` extension

Check how to import it to Z2M

https://www.zigbee2mqtt.io/advanced/more/external_converters.html

```javascript
import * as fz from "zigbee-herdsman-converters/converters/fromZigbee";
import * as tz from "zigbee-herdsman-converters/converters/toZigbee";
import * as exposes from "zigbee-herdsman-converters/lib/exposes";
import * as reporting from "zigbee-herdsman-converters/lib/reporting";

const e = exposes.presets;

export default {
    fingerprint: [
        {modelID: "ON/OFF(2CH)", softwareBuildID: "2.5.3_r2"},
        {modelID: "ON/OFF(2CH)", softwareBuildID: "2.9.2_r3"},
        {modelID: "ON/OFF(2CH)", softwareBuildID: "2.9.2_r5"},
        {modelID: "ON/OFF(2CH)", softwareBuildID: "2.9.2_r54"},
    ],
    model: "SR-ZG9101SAC-HP-SWITCH-2CH",
    vendor: "Sunricher",
    description: "Zigbee 2 channels switch",
    fromZigbee: [fz.on_off, fz.electrical_measurement, fz.metering, fz.power_on_behavior, fz.ignore_genOta],
    toZigbee: [tz.on_off, tz.power_on_behavior],
    exposes: [
        e.switch().withEndpoint("l1"),
        e.switch().withEndpoint("l2"),
        e.power(),
        e.current(),
        e.voltage(),
        e.energy(),
        e.power_on_behavior(["off", "on", "previous"]),
    ],
    meta: {multiEndpoint: true, multiEndpointSkip: ["power", "energy", "current", "voltage"]},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint1 = device.getEndpoint(1);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint1, coordinatorEndpoint, ["genOnOff"]);
        await reporting.bind(endpoint2, coordinatorEndpoint, ["genOnOff"]);
        await reporting.onOff(endpoint1);
        await reporting.onOff(endpoint2);
        if (
            (device && device.softwareBuildID === "2.9.2_r3")
            (device && device.softwareBuildID === "2.9.2_r5")
            (device && device.softwareBuildID === "2.9.2_r54")
        ) {
            // newer firmware version power reports are on endpoint 11
            const endpoint11 = device.getEndpoint(11);
            await reporting.bind(endpoint11, coordinatorEndpoint, ["haElectricalMeasurement", "seMetering"]);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint11);
            await reporting.activePower(endpoint11);
            await reporting.rmsCurrent(endpoint11, {min: 10, change: 10});
            await reporting.rmsVoltage(endpoint11, {min: 10});
            await reporting.readMeteringMultiplierDivisor(endpoint11);
        } else if (device && device.softwareBuildID === "2.5.3_r2") {
            // older firmware version power reports are on endpoint 1
            await reporting.bind(endpoint1, coordinatorEndpoint, ["haElectricalMeasurement", "seMetering"]);
            await reporting.readEletricalMeasurementMultiplierDivisors(endpoint1);
            await reporting.activePower(endpoint1);
            await reporting.rmsCurrent(endpoint1, {min: 10, change: 10});
            await reporting.rmsVoltage(endpoint1, {min: 10});
            await reporting.readMeteringMultiplierDivisor(endpoint1);
            await reporting.currentSummDelivered(endpoint1);
        }
    },
};
```

chris-1243, 2025-08-28:
You will also have to reconfigure your device by pressing on the yellow square button