Koenkk/zigbee2mqtt #28333: Can't pairing Schneider Electric CCT5011-0002

juliannideltorri, 2025-08-20:
### What happened?

Hello,

I’ve been using a Schneider Electric CCT5011-0002 for a few years now.
After upgrading from zigbee2mqtt 1.8 to the latest version, I’m no longer able to pair this device.

I’ve tried pairing without an install code, then with the install code, but I don’t see any trace of the device in the logs.

I’ve seen that other users have reported a similar issue — is there a known fix or workaround?
https://community.home-assistant.io/t/cannot-pair-switch-schneider-electric-cct5011-0002w-to-my-zigbee2mqtt-network/493224

Thanks in advance for your help.

### Zigbee2MQTT version

 2.6.0 

### Adapter firmware version

 20201026

### Adapter

Electrolama zzh!

### Setup

HomeAssistant on rasberryPi 4

juliannideltorri, 2025-10-26:
Hi, I resolved problem to adding coordinator (smart plug Xiaomi).
I don't now why, but it works.

closed 2025-10-26 by juliannideltorri

sorano, 2025-10-26:
> Hi, I resolved problem to adding coordinator (smart plug Xiaomi). I don't now why, but it works.

Could you expand on what you mean with:  "I resolved problem to adding coordinator (smart plug Xiaomi)" ?