Koenkk/zigbee2mqtt #28336: [Feature request]: merge `device_options` recursively

jackwilsdon, 2025-08-20:
### Is your feature request related to a problem? Please describe

I have a global `device_options.homeassistant` which prevents a number of values from being sent to Home Assistant:
```yaml
device_options:
  homeassistant:
    color_options: null
    device_mode: null
    effect: null
    energy: null
    identify: null
    led_indication: null
    linkquality: null
    motion_sensitivity: null
    occupancy_timeout: null
    power_on_behavior: null
    switch_led_indication: null
    update_available: null
    update_state: null
    voltage: null
```

When I want to disable any additional values on a per-device basis, I have to re-define these values:

```
  "0x0000000000000000":
    friendly_name: Kitchen Ceiling Light 1
    # We have to include all of the values for this device as device_options
    # is merged shallowly.
    homeassistant:
      effect: null
      # All I want to disable is "light" (as this light is in a group).
      light: null
      linkquality: null
      power_on_behavior: null
      update_available: null
      update_state: null
```

### Describe the solution you'd like

`device_options` should be merged recursively with options defined under `devices.<id>`.

### Describe alternatives you've considered

Re-defining all applicable options on each device.

### Additional context

None

Nerivec, 2025-08-24:
https://github.com/Koenkk/zigbee2mqtt/blob/bda5bf9e966278c6d25483221130c0f8879d0a5a/lib/model/device.ts#L28-L31

@Koenkk it's a bugfix really, but technically ends up a breaking change (could result in nasty behavior for some users). We'd have to put it under a toggle (like `advanced.legacy_override_nested_options` - default: true) for now?

Koenkk, 2025-08-25:
I'm not sure if its a bug fix, currently only the root is merged (which is expected for now).

Can't [YAML anchors](https://support.atlassian.com/bitbucket-cloud/docs/yaml-anchors/) be used for this?

amosyuen, 2025-10-29:
I'm also very interested in this feature.

I tried using YAML anchors, but this doesn't work so well because whenever the yaml file is changed by zigbee2mqtt (config migration, device added/removed, setting changed in UI) then it overwrites the configuration file with the config file with all anchors merged in.

I then need to go back and figure out the diffs, revert back to my original file, and apply the new changes. My config file has 113 devices, 2k+ lines so this can be quite time consuming.