Koenkk/zigbee2mqtt #28342: [New device support]: ZX-8588-675 Luminea Home Control soil Humidity Sensor

dragonlost, 2025-08-20:
### Link

https://www.pearl.fr/article/ZX8588/thermometre-hygrometre-de-sol-connecte-compatible-zigbee

### Database entry

{"id":65,"type":"EndDevice","ieeeAddr":"0xa4c138c90233d845","nwkAddr":45628,"manufId":4417,"manufName":"_TZE200_9cqcpkgb","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"݄70iބ70i߄70i�70i�70i�70i�70i�70i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":72,"modelId":"TS0601","manufacturerName":"_TZE200_9cqcpkgb","powerSource":3,"zclVersion":3,"hwVersion":1}},"genGroups":{"attributes":{"nameSupport":0}},"genScenes":{"attributes":{"count":0,"currentGroup":0,"currentScene":0,"nameSupport":0,"sceneValid":0}}},"binds":[{"cluster":25,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe522f0c","endpointID":1},{"cluster":10,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe522f0c","endpointID":1},{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe522f0c","endpointID":1},{"cluster":4,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe522f0c","endpointID":1},{"cluster":61184,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe522f0c","endpointID":1},{"cluster":5,"type":"endpoint","deviceIeeeAddress":"0x8c65a3fffe522f0c","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":72,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1755700603849}

### Zigbee2MQTT version

2.6.0

### Comments

Hello everyone.

I received a new Zigbee device today.
I previously purchased this device through AliExpress. Generally, it's detected without any problems by Z2M.
I purchased this model through Pearl, and the signature is no longer the same; therefore, it is considered incompatible by Z2M.

Z2m version : 2.6.0, zigbee-herdsman-converters : 24.11.0, zigbee-herdsman : 5.0.4

I tried this converter, it loads but the values remain on "N.A":
```

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: "TS0601",
            manufactureName: "_TZE200_9cqcpkgb",
        },
    ],
    model: "TS0601_soil",
    vendor: "Tuya",
    description: "Soil sensor",
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.temperature(),
        e.soil_moisture(),
        tuya.exposes.temperatureUnit(), 
        e.battery(), 
        tuya.exposes.batteryState()],
    meta: {
        tuyaDatapoints: [
            [3, "soil_moisture", tuya.valueConverter.raw],
            [5, "temperature", tuya.valueConverter.raw],
            [9, "temperature_unit", tuya.valueConverter.temperatureUnit],
            [14, "battery_state", tuya.valueConverter.batteryState],
            [15, "battery", tuya.valueConverter.raw],
        ],
    },
    extend: [],
};

module.exports = definition;
```

I used the tuya converter for this model (used on my other aliexpress models): https://www.zigbee2mqtt.io/devices/TS0601_soil.html

### External definition

```shell
export default {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_9cqcpkgb',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};
```

### What does/doesn't work with the external definition?

No parameter uploaded works

dragonlost, 2025-08-20:
Here is a debug report from Z2M:

```

[2025-08-20 16:36:38] [34mdebug[39m: 	z2m: Received Zigbee message from 'sol_potager', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
[2025-08-20 16:36:38] [34mdebug[39m: 	z2m: Skipping message, still interviewing
[2025-08-20 16:36:38] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sol_potager', payload '{"last_seen":"2025-08-20T14:36:38.280Z","linkquality":156,"temperature_unit":"celsius"}'
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:controller:device: Interview - got 'zclVersion' for device '0xa4c138c90233d845'
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:controller:endpoint: ZCL command 0xa4c138c90233d845/1 genBasic.read(["appVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolicy":"immediate"})
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember: ~~~> [ZCL to=0xa4c138c90233d845:45628 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":11,"commandIdentifier":0}]
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=62 Len=27]
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=6 frmRx=6](ackRx=6)
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7](ackRx=6 frmTx=7)
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=6](frmRx=6) Added to rxQueue
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=7)
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=62 Len=10]
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=80 messageTag=25 status=OK]
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7](ackRx=7 frmTx=7)
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=7](frmRx=7) Added to rxQueue
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=7)
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=62 Len=26]
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:ezsp: ezspMessageSentHandler: status=OK type=DIRECT indexOrDestination=45628 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":80} messageTag=25
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7](ackRx=7 frmTx=7)
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=0](frmRx=0) Added to rxQueue
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=1](ackRx=7)
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=62 Len=45]
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:ember:ezsp: ezspIncomingMessageHandler: type=UNICAST apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":320,"groupId":0,"sequence":231} packetInfo:{"senderShortId":45628,"senderLongId":"0x0000000000000000","bindingIndex":255,"addressIndex":33,"lastHopLqi":156,"lastHopRssi":-72,"lastHopTimestamp":84565132} messageContents=180b010100002048
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:controller: Received payload: clusterID=0, address=45628, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=156, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":11,"commandIdentifier":1},"payload":[{"attrId":1,"status":0,"dataType":32,"attrData":72}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"statusEquals","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"statusEquals","value":0}]}]}}
[2025-08-20 16:36:38] [34mdebug[39m: 	z2m: Received Zigbee message from 'sol_potager', type 'readResponse', cluster 'genBasic', data '{"appVersion":72}' from endpoint 1 with groupID 0
[2025-08-20 16:36:38] [34mdebug[39m: 	z2m: Skipping message, still interviewing
[2025-08-20 16:36:38] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/sol_potager', payload '{"last_seen":"2025-08-20T14:36:38.388Z","linkquality":156,"temperature_unit":"celsius"}'
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:controller:device: Interview - got 'applicationVersion' for device '0xa4c138c90233d845'
[2025-08-20 16:36:38] [34mdebug[39m: 	zh:controller:endpoint: ZCL command 0xa4c138c90233d845/1 genBasic.read(["stackVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false,"sendPolic [... cut ...] nse":true,"direction":0,"reservedBits":0,"transactionSequenceNumber":249,"writeUndiv":false})
[2025-08-20 16:36:49] [34mdebug[39m: 	zh:ember: ~~~> [ZCL to=0xa4c13816d0159ff0:26909 apsFrame={"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":249,"commandIdentifier":11}]

```

dragonlost, 2025-08-22:
I checked via a Tuya gateway and the tutorial on the Z2M website.
The numbers and data format are indeed the same.

3, "soil_moisture" 
5, "temperature"
9, "temperature_unit"
14, "battery_state"
15, "battery"

Koenkk, 2025-08-24:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-08-24 by Koenkk

dragonlost, 2025-08-24:
Thanks for the change !

dragonlost, 2025-09-10:
an example of the same transaction with another ground sensor of the same type.

`[2025-09-10 20:37:16] �[34mdebug�[39m: 	zh:controller: Received payload: clusterID=0, address=8735, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=104, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":9,"commandIdentifier":1},"payload":[{"attrId":4,"status":0,"dataType":66,"attrData":"_TZE200_9cqcpkgb"},{"attrId":0,"status":0,"dataType":32,"attrData":3},{"attrId":1,"status":0,"dataType":32,"attrData":72},{"attrId":5,"status":0,"dataType":66,"attrData":"TS0601"},{"attrId":7,"status":0,"dataType":48,"attrData":3},{"attrId":65534,"status":0,"dataType":48,"attrData":0}],"command":{"ID":1,"name":"readRsp","parameters":[{"name":"attrId","type":33},{"name":"status","type":32},{"name":"dataType","type":32,"conditions":[{"type":"fieldEquals","field":"status","value":0}]},{"name":"attrData","type":1000,"conditions":[{"type":"fieldEquals","field":"status","value":0}]}]}} 
[2025-09-10 20:37:16] �[34mdebug�[39m: 	z2m: Received Zigbee message from 'sol_potager', type 'readResponse', cluster 'genBasic', data '{"65534":0,"appVersion":72,"manufacturerName":"_TZE200_9cqcpkgb","modelId":"TS0601","powerSource":3,"zclVersion":3}' from endpoint 1 with groupID 0 
`