Koenkk/zigbee2mqtt #28352: [Wrong device]: Paulmann 291.46 being picked up as Paulmann 291.52

Zarn-Smith, 2025-08-21:
### Link

https://uk.paulmann.com/p/230-v-standard-smart-home-zigbee-3.0-led-candle-e14-470lm-5w-rgbw-dimmable-matt/29146

### Model

291.46

### Description

Smart Home Zigbee 3.0 LED Candle E14 470lm 5W RGBW+ Dimmable

### Vendor

Paulmann

### Picture (link)

https://content.paulmann.com/media/f8/b4/09/1694182468/29146fr01.jpg?ts=1755768080&width=1160&quality=80

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0xa4c13800ac4a2d8e","nwkAddr":19626,"manufId":4632,"manufName":"Paulmann Licht GmbH","powerSource":"Mains (single phase)","modelId":"RGBWW","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,4111],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"RGBWW","manufacturerName":"Paulmann Licht GmbH","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":112,"swBuildId":"PIIC5804","productUrl":"www.paulmann.com"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":31,"colorTempPhysicalMin":153,"colorTempPhysicalMax":454,"currentHue":35,"colorMode":0,"enhancedCurrentHue":0,"colorTemperature":412,"startUpColorTemperature":370}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{"onLevelSupported":false}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":112,"swBuildId":"PIIC5804","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1755777647171}

### Notes

Everything seems to be working ok with the bulb, it is just being picked up as a 291.52 instead of a 291.46 bulb.

CaeruleusAqua, 2025-09-20:
Unfortunately, I cannot confirm that everything is working correctly. Switching it on and off does not work reliably. Sometimes it does not turn on at all. However, it can then be switched on using relative dimming. When the lamp is switched on using relative dimming, it turns on, but the on/off status is not updated. After that, the lamp can no longer be switched on or off until the brightness or color has been set to an absolute value.

Zarn-Smith, 2025-09-20:
@CaeruleusAqua - I don't seem to be suffering the challenges that you are facing. If you could describe the steps you are taking, I can try to repeat to confirm if there is a problem?

CaeruleusAqua, 2025-09-21:
Unfortunately, I'm currently on a business trip. I'll be back on Thursday, then I can go through it again.

absent42, 2025-10-17:
I have a 291.50 which is also being detected as a 291.52.  @Zarn-Smith could you check that your bulb is being correctly identified with the below external converter?

```
import * as m from "zigbee-herdsman-converters/lib/modernExtend";

export default {
    zigbeeModel: ["RGBWW"],
    model: "291.52",
    vendor: "Paulmann",
    description: "Smart Home Zigbee LED bulb 4,9W Matt E14 RGBW",
    extend: [m.light({colorTemp: {range: [153, 454]}, color: {modes: ["xy", "hs"]}})],
    whiteLabel: [
        {
            model: "291.50",
            vendor: "Paulmann",
            description: "230 V Standard Smart Home Zigbee 3.0 LED Drop E14 470lm 5W RGBW+ dimmable Matt",
            fingerprint: [{model: "291.50"}],
        },
        {
            model: "291.46",
            vendor: "Paulmann",
            description: "230 V Standard Smart Home Zigbee 3.0 LED Candle E14 470lm 5W RGBW+ dimmable Matt",
            fingerprint: [{model: "291.46"}],
        },
    ],
    meta: {},
};
```

absent42, 2025-10-18:
In my searches Paulmann 29152 seems to be a 3-pack of Paulmann 29146 https://uk.paulmann.com/p/230-v-standard-smart-home-zigbee-3.0-led-candle-e14-3x470lm-3x5w-rgbw-dimmable-matt/29152 and the image that shows up as 291.52 in Zigbee2MQTT seems to actually be the top of a Solar Lamp Post Light https://www.amazon.co.uk/Kemeco-ST6261SS4-Exterior-Outdoor-Landscape/dp/B08XX67J9J

Zarn-Smith, 2025-10-19:
Ah yes. I never noticed that. So to fix the original converter all that would be needed is to update the picture and change the reported model number.

closed 2025-12-27 by github-actions[bot]