Koenkk/zigbee2mqtt #28402: Z2M stops working

surfingbytes, 2025-08-27:
### What happened?

I have SLZB-06Mg24 as coordinator, connected via LAN. I'm not sure if the issue is there or in Z2M, but sometimes (usually when I build and update non-related docker on my server where Z2M is running), all my zigbee devices goes offline. SLZB is still running, and in Z2M logs, I see this:
[2025-08-28 00:28:55] error: zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
[2025-08-28 00:28:55] error: zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-08-28 00:28:55] error: zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-08-28 00:28:55] error: zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-08-28 00:28:55] error: z2m: Adapter disconnected, stopping

After there logs, everything fails to ping and everything is offline:

[2025-08-28 00:29:42] warning: z2m: Failed to ping 'Bathroom Fan' (attempt 1/2, ZCL command 0x842e14fffe0fd45d/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL to=0x842e14fffe0fd45d:5764 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=NETWORK_DOWN.))
[2025-08-28 00:29:48] warning: z2m: Failed to ping 'Bathroom Fan' (attempt 2/2, ZCL command 0x842e14fffe0fd45d/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL to=0x842e14fffe0fd45d:5764 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=NETWORK_DOWN.))
[2025-08-28 00:30:26] warning: z2m: Failed to ping 'Living Room Lights' (attempt 1/2, ZCL command 0x5c0272fffed9e3b7/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL to=0x5c0272fffed9e3b7:25393 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=NETWORK_DOWN.))
[2025-08-28 00:30:32] warning: z2m: Failed to ping 'Living Room Lights' (attempt 2/2, ZCL command 0x5c0272fffed9e3b7/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL to=0x5c0272fffed9e3b7:25393 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=NETWORK_DOWN.))
[2025-08-28 00:30:37] warning: z2m: Failed to ping 'Living Room IR Blaster Top' (attempt 1/2, ZCL command 0xf84477fffe519a7a/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL to=0xf84477fffe519a7a:6994 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=NETWORK_DOWN.))
[2025-08-28 00:30:43] warning: z2m: Failed to ping 'Living Room IR Blaster Top' (attempt 2/2, ZCL command 0xf84477fffe519a7a/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL to=0xf84477fffe519a7a:6994 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=NETWORK_DOWN.))
[2025-08-28 00:31:25] warning: z2m: Failed to ping 'Living Room Back Light Left' (attempt 1/2, ZCL command 0x60a423fffef15009/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (~x~> [ZCL to=0x60a423fffef15009:17699 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0}] Failed to send request with status=NETWORK_DOWN.))

Everything goes back online when I restart Z2M docker, which, to me, means there is something wrong with Z2M. Even if the original issue is elsewhere, Z2M should auto-recover from this (even if it means it restarts itself), and not wait until the docker is manually restarted. If Z2M exits on this error, docker (with restart: unless-stopped) would restart it and it would recover automatically. No point of keeping running, but failing to actually work..

### What did you expect to happen?

As restart of Z2M helps, it means Z2M could either recover, or exit to have docker engine restart it.

### Zigbee2MQTT version

2.6.0 commit: 0bdc3fd1f76aaad71f6231e5b38529e2d2b345e9 

### Adapter firmware version

8.0.2 [GA]

### Adapter

EmberZNet - SLZB-06Mg24

### Setup

Docker

Grufbert, 2025-09-06:
My system was experiencing not responding Zigbee devices and showed some ZCL errors. Adding the `Z2M_WATCHDOG=default` parameter to docker-compose configuration resolved the stability issues.

Thanks @francisp2 for the hint!

Error call stack:
```
[2025-09-06 09:19:47] debug: 	z2m: Error: ZCL command 0x847217fffeccf377/1 lightingColorCtrl.moveToColorTemp({"colortemp":436,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC no ack' (233))
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:541:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:639:20)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:639:20)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:639:20)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:639:20)
    at /app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:439:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:437:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:86:25)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:358:20)
```

bskjon, 2025-10-04:
I have a similar issue, but rather midler.
Some of my devices randomly enters "Offline" while nearby devices remain Online and responsive. 
Nothing makes the offline device respond, and simply requesting Z2M to restart makes all Offline devices go online and responsive again.