Koenkk/zigbee2mqtt #28424: Missing support for Zigbee Green Power command 55 (double press) for Legrand ZLGP17/ZLGP18 in Zigbee2MQTT and Home Assistant

dasimon135, 2025-08-31:
### What happened?

When performing a double press on my Legrand ZLGP17/ZLGP18 device, Zigbee2MQTT reports a missing or unsupported command 55. The device does not respond to this double press action in Home Assistant, leading to errors or a lack of expected behavior.

### What did you expect to happen?

I expected the double press action (command 55) to be recognized and processed correctly by Zigbee2MQTT and reflected in Home Assistant, enabling the automation or manual control mapped to this action to work properly.

### How to reproduce it (minimal and precise)

1. Include a Legrand ZLGP17 or ZLGP18 device in Zigbee2MQTT integrated with Home Assistant.
2. Perform a double press on the physical button of the device.
3. Observe that command 55 is triggered but not supported or correctly handled, resulting in error logs or no action triggered in Home Assistant.

### Zigbee2MQTT version

2.6.0

### Adapter firmware version

8.0.2 [GA]

### Adapter

EmberZNet  

### Setup

I run Zigbee2MQTT as a Docker container on a Synology NAS (x86-64 architecture)

chris-1243, 2025-08-31:
It should be solved with

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9896

The fix might be available in the next official release or in the latest-dev when accepeted and merged by the dev-team

Koenkk, 2025-09-02:
Available in the 2.6.1 release.

closed 2025-09-02 by Koenkk

dasimon135, 2025-09-02:
> Available in the 2.6.1 release.
Hello, good news! Installed today and it works well... Except it's reversed, top is bottom and bottom is top 🤔