Koenkk/zigbee2mqtt #28445: Changing Zigbee channel resets network key (network key from configuration.yaml reset)

dane-io, 2025-09-02:
### What happened?

Hello,

I just updated to 2.6.1 (skipped 2.6.0 due to frontend bug that made it impossible to change Zigbee settings via the frontend on Home Assistant). After attempting to change the Zigbee channel, I found that the network_key in is reset in configuration.yaml. This causes Z2M to be unable to start.

Upon further investigation, it looks like the network_key value is not imported into the settings page, thus to change any setting without breaking the network I would have to manually enter the network key. It is unclear to me if this is an issue with the frontend or Z2M itself.

### What did you expect to happen?

My network_key should be imported to the settings page from configuration.yaml and respected when I change other Zigbee settings.

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

20210708

### Adapter

SONOFF Zigbee Dongle

### Setup

Zigbee2MQTT add-on in Home Assistant running in Proxmox VM

enigmaspb, 2025-09-02:
Same after enable "last seen" in webui , lost network key in configuration.yaml

[2025-09-02 06:51:17] error:   zh:adapter:zstack:manager: - Network Key: configured=01030507090b0d0f00020406080a0c0d, adapter:active=067439ddeaf4b6139f27e098712f9bdf adapter:preconfigured=067439ddeaf4b6139f27e098712f9bdf, adapter:alternate=067439ddeaf4b6139f27e098712f9bdf

ImrirS, 2025-09-03:
When changing the logging level from info to debug, the same thing happens.

rolandjutai, 2025-09-13:
Yes. Annoying bug. Can't use the settings page so using the yaml to configure things... :/