Koenkk/zigbee2mqtt #28446: Sonoff ZBTRV

siggino12, 2025-09-02:
### What happened?

Hey,

In Zigbee2mqtt its show me the runningstate of the valve.
But in Mqtt not.
How can resolve that ?
Maybe somebody have idea or maybe need update ?

Thank you so much

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

20240710

### Adapter

Snzb 06p7

### Setup

Vmwar on nas synology

siggino12, 2025-09-02:
🙏🏻🙏🏻🙏🏻🙏🏻🙏🏻🙏🏻🙏🏻

GigiPompieru, 2025-09-02:
In MQTT there is a climate.device entity that has multiple states assigned to it, one of the states is what you are looking for: ```hvac_action```. This changes depending on what the TRV is doing: ```heating``` or ```idle```.

siggino12, 2025-09-02:
Hey 

> In MQTT there is a `climate.device` entity that has multiple states assigned to it, one of the states is what you are looking for: `hvac_action`. This changes depending on what the TRV is doing: `heating` or `idle`.

H

closed 2025-09-02 by siggino12