Koenkk/zigbee2mqtt #28450: Lutron Dimmer Switch Z3-1BRL not turning on lights anymore after update to 2.6.1

jkyselak, 2025-09-02:
### What happened?

Lutron Dimmer Switch Z3-1BRL not turning on lights anymore after update to 2.6.1. Once Lights have been turned on manually, they can be turned off with the switch but not on. This is the automation being used:

alias: Lutron Aurora Dimmer Dining Room
description: MQTT trigger for Lutron Aurora Dimmer Knob
triggers:
  - trigger: mqtt
    topic: zigbee2mqtt/Dining Room Light Switch
    payload: brightness_move_to_level
    value_template: "{{ value_json.action }}"
actions:
  - choose:
      - conditions:
          - condition: template
            value_template: "{{ action_level == 0 }}"
        sequence:
          - data:
              transition: "{{ transition_time }}"
            target:
              device_id: 10dee8117f180286ca050fdb8a9fbc66
            action: light.turn_off
    default:
      - target:
          device_id: 10dee8117f180286ca050fdb8a9fbc66
        data:
          transition: "{{ transition_time }}"
          brightness: "{{ action_level }}"
        action: light.turn_on
mode: queued
max_exceeded: silent
variables:
  action: "{{ trigger.payload_json.action }}"
  action_level: "{{ trigger.payload_json.action_level }}"
  transition_time: "{{ trigger.payload_json.action_transition_time }}"

### What did you expect to happen?

To be able to turn lights off and on with the Z3-1BRL

### How to reproduce it (minimal and precise)

Press Z3-1BRL

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

20240710

### Adapter

tubezb-cc2652p2-ethusb-2022

### Setup

Add-on on Home Assistant OS on Asus PL64

### Device `database.db` entry

{"id":40,"type":"EndDevice","ieeeAddr":"0xffff000fe7fc05e7","nwkAddr":24018,"manufId":4420,"manufName":"Lutron","powerSource":"Battery","modelId":"Z3-1BRL","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":2080,"inClusterList":[0,1,3,4096,64512],"outClusterList":[3,4,6,8,25,4096],"clusters":{"genBasic":{"attributes":{"modelId":"Z3-1BRL","manufacturerName":"Lutron","powerSource":3,"zclVersion":3,"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20181019","swBuildId":"3.8"}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ebefd","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b002a2ebefd","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":1,"stackVersion":1,"hwVersion":1,"dateCode":"20181019","swBuildId":"3.8","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1756802832345}

deviantintegral, 2025-09-03:
In my HA automation:

> Error: In 'template' condition: TypeError: '>' not supported between instances of 'NoneType' and 'int'

Here's a message in 2.6.1. Note how `action_level` is `null` when clicking the button:

> zigbee2mqtt  | [2025-09-02 21:12:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Family Room Dial', payload '{"action":"brightness_move_to_level","action_level":null,"action_transition_time":0.07,"battery":null,"brightness":null,"last_seen":"2025-09-03T01:12:06.439Z","linkquality":123,"source_route":[62119,47927],"update":{"installed_version":3080,"latest_version":3080,"state":"idle"}}'

In 2.6.0 `action_level` is `255` which clicking the main button:

> zigbee2mqtt  | [2025-09-02 21:14:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Family Room Dial', payload '{"action":"brightness_move_to_level","action_group":12269,"action_level":255,"action_transition_time":0.07,"battery":null,"brightness":null,"last_seen":"2025-09-03T01:14:27.952Z","linkquality":127,"source_route":[62119,47927],"update":{"installed_version":3080,"latest_version":3080,"state":"idle"}}'

Nerivec, 2025-09-03:
I'm wondering if this is intended (expect to receive special handling or something)? I'd be surprised if spec wasn't followed on a good-old uint8...
I'm not exactly sure how this device works though? How does this one level value affect the entire on/off thing? I see `action_group` is also missing in the payload?

deviantintegral, 2025-09-04:
Probably more details than needed but...

The device has two controls:

- You can press the center button as a toggle for on / off.
- You can turn the outside of it as a dial to control brightness.

However, the center button is weird. It only sends level control messages, and not on-off messages. So if it's turning devices on, it sends 255, and 0 for off.

It also supports direct binding but only to individual devices, not groups. Yet, I see it sending to a group ID that I don't think z2m knows about? Is it possible for a device to create a group without the coordinator knowing about it?

```
Frame 6416: 52 bytes on wire (416 bits), 52 bytes captured (416 bits)
IEEE 802.15.4 Data, Src: 0xfaf2, Dst: Broadcast
ZigBee Network Layer Data, Dst: Broadcast, Src: 0x1d90
    Frame Control Field: 0x0248, Frame Type: Data, Discover Route: Enable, Security Data
    Destination: 0xfffd
    Source: 0x1d90
    Radius: 8
    Sequence Number: 247
    [Extended Source: ff:ff:00:0f:e7:fb:37:e6 (ff:ff:00:0f:e7:fb:37:e6)]
    [Origin: 6034]
    ZigBee Security Header
ZigBee Application Support Layer Data, Group: 0x2fed, Src Endpt: 1
    Frame Control Field: Data (0x0c)
    Group: 0x2fed
    Cluster: Level Control (0x0008)
    Profile: Home Automation (0x0104)
    Source Endpoint: 1
    Counter: 71
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 0
    Command: Move to Level with OnOff (0x04)
    Payload
        Level: 255
        Transition Time: 0.7 seconds
```

Koenkk, 2025-09-04:
I fixed the converter (`null` will be mapped to `255` now)

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Nerivec, 2025-09-04:
It wouldn't be null in the converter, it would be NaN, wouldn't it? That's before it's stringified.

Koenkk, 2025-09-04:
> It wouldn't be null in the converter, it would be NaN, wouldn't it? That's before it's stringified.

good point, fixed it properly now

LowFatMom, 2025-10-21:
I have a bunch of these and I always experienced problems with them when 4x Z3 are direct bound to the same x8 bulbs. 

It works fine for a day then the dimming controls are randomly lost (ie, one dimmer would start only controlling 4 of the 8 lights via the dimmer, but they always respond to the center on/off button) 

Each dimmer would randomly lose certain bulbs. Could this be related to some groups conflict ? Just creating that group id 12262 would fix this? 

The area where only one Lutron controls a group don’t have this issue.

LowFatMom, 2025-10-22:
> > ie, one dimmer would start only controlling 4 of the 8 lights via the dimmer, but they always respond to the center on/off button)
> 
> That makes it seem like it's not losing the group association, otherwise on / off wouldn't work either.
> 
> > The area where only one Lutron controls a group don’t have this issue.
> 
> I updated the docs for this at https://www.zigbee2mqtt.io/devices/Z3-1BRL.html#device-binding. Can you see if your dials are using different group IDs or not? Now that I think of it, I wonder if they choose a random ID or one based on the unique device, otherwise you could never use more than one dial on the same network to control different lights.

alright, not sure if I did this right, is that it? 

[2025-10-22 07:59:17] debug:    z2m: No converter available for '9290022267A' with cluster 'genGroups' and type 'commandGetMembershipRsp' and data '{"capacity":24,"groupcount":1,"grouplist":[19056]}'

deviantintegral, 2025-10-26:
Good news, I found my sniffer.

Bad news: while the dimmer uses groups for on / off, it sends device specific commands for dimming. I'm surprised it is as smooth as it is controlling 6 different bulbs! But, that means manually adding devices to the group will only work for on / off.

LowFatMom, 2025-10-26:
Thx for the update ! Sadly it doesn’t looks like there is a fix for my multi dimmers/room issue that seems to lose dimming bindings over time.