Koenkk/zigbee2mqtt #28451: Tuya TS0203 (Door Sensor)was found as iHseno ZC-W1(Temperature Sensor) after Update

nineteen0815, 2025-09-02:
### Link

https://www.zigbee2mqtt.io/devices/TS0203.html

### Model

unknown

### Description

Door/window sensor

### Vendor

Tuya

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/TS0203.png

### Database entry

0xa4c138f10b55deea , ZC-W1

maz0r, 2025-09-02:
Related to this #28167

maz0r, 2025-09-02:
Here's one of mine
```json
{"id":62,"type":"EndDevice","ieeeAddr":"0xa4c13808bf7b0644","nwkAddr":12583,"manufId":4742,"manufName":"Zbeacon","powerSource":"Battery","modelId":"TS0203","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0203","manufacturerName":"Zbeacon","powerSource":3,"zclVersion":3,"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"ssIasZone":{"attributes":{"61441":2,"iasCieAddr":"0x00124b0031f90b62","zoneState":1,"currentZoneSensitivityLevel":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":146,"batteryVoltage":26}},"genPollCtrl":{"attributes":{"checkinInterval":6960}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0031f90b62","endpointID":1},{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0031f90b62","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1371223903},"lastSeen":1756836209541,"checkinInterval":1740}
```

Koenkk, 2025-09-02:
Fixed in https://github.com/Koenkk/zigbee2mqtt/issues/28167

If you **do not want to update** you can apply the following workaround:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `Zbeacon` to `Zbeacon_ignore`
- Start Z2M

xavbart, 2025-09-02:
Confirmed working on Zigbee2MQTT-Edge 
(just received those sensors, was testing with latest Zigbee2MQTT release, got this "iHseno Temperature Sensor" instead of my expected Tuya door sensors from Aliexpress, but found this discussion and installed Edge, all is sorted!) 
Well done

doming52, 2025-09-03:
It worked! Thanks for your support.

davinkevin, 2025-09-22:
Got a few of them and I'm waiting for the next release, it should be soon no? 😇

janniklas2501-coder, 2025-09-25:
> Fixed in [#28167](https://github.com/Koenkk/zigbee2mqtt/issues/28167)
> 
> If you **do not want to update** you can apply the following workaround:
> 
> * Stop Z2M
> * Open `data/database.db` with a text editor and change all occurrences of `Zbeacon` to `Zbeacon_ignore`
> * Start Z2M

Worked for me, then i had to restart the server and now its not recognized right altough i have Zbeacon_ignore in the data