Koenkk/zigbee2mqtt #28455: Lot of errors from Tuya PJ-1203A

syntesys87, 2025-09-02:
### What happened?

From last update, I notice a lot of errors from Tuya PJ-1203A:

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

8.0.2

### Adapter

Sonoff ZBDongle-E (ZigBee 3.0 USB Dongle Plus) – v2

### Setup

Z2M on Raspberry PI 5 16GB, HAOS

### Debug log

[log.log](https://github.com/user-attachments/files/22097128/log.log)

  '0xa4c138d8100e1a0a':
    friendly_name: Consumo Clima Induzione
    signed_power_b: true
    signed_power_a: true
    ac_frequency_calibration: 1
    homeassistant: {}
    optimistic: true
    qos: 0
  '0xa4c13854bfbb1f58':
    friendly_name: Consumo Casa Fotovoltaico
    ac_frequency_calibration: 1
    homeassistant: {}
    optimistic: true
    qos: 0
  '0xa4c138554cc92b21':
    friendly_name: Consumo Forno Microonde
    signed_power_a: true
    signed_power_b: true
    ac_frequency_calibration: 1
    homeassistant: {}
    optimistic: true
    qos: 0
  '0xa4c13831d78ec9a7':
    friendly_name: Consumo Frigo Cantina
    signed_power_a: true
    signed_power_b: true
    ac_frequency_calibration: 1
    homeassistant: {}
    optimistic: true
    qos: 0  
'0xa4c1389bf2e86240':
    friendly_name: Consumo UPS
    ac_frequency_calibration: 1
    homeassistant: {}
    optimistic: true
    qos: 0

mnmarc, 2025-09-02:
I‘ve got also errors. Since updating Zigbee2MQTT Add-on from v2.6.0-1 to v2.6.1-1 on Home Assistant OS,
my PJ-1203A (`_TZE204_81yrt3lo`, 0xa4c138c6e459e106) started showing continuous time sync errors:

zhc:tuya: Failed to sync time with ‘0xa4c138c6e459e106’
(Error: ZCL command manuSpecificTuya.mcuSyncTime failed (Delivery failed for ‘42769’.))

With v2.6.0-1 this did not happen.  

Not sure if this is expected behavior after the recent timestamp improvements –  
but maybe the PJ-1203A does not support `mcuSyncTime` correctly?

Environment:
- Home Assistant OS (latest)
- Zigbee2MQTT Add-on v2.6.1-1
- Sonoff Dongle-E (latest)

Koenkk, 2025-09-02:
Could you provide the debug log of this?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

PeetersJan, 2025-09-03:
Same issue here.  I will post a log tonight.

[image]

Koenkk, 2025-09-03:
Fixed!

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-09-03 by Koenkk

Koenkk, 2025-09-05:
Also fixed that one.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

JNow01, 2025-09-06:
Hello,
I'm sorry but I also still get loads of "mcuSyncTime" errors with the Dev version and 3 out of 5 [Cafago valves](https://www.zigbee2mqtt.io/devices/TS0601_thermostat_4.html#tuya-ts0601_thermostat_4) :
```
[2025-09-06 16:41:18] error: 	zhc:tuya: Failed to sync time with '0xa4c1380b4e2269c9' (Error: ZCL command 0xa4c1380b4e2269c9/1 manuSpecificTuya.mcuSyncTime({"payloadSize":8,"payload":[104,188,72,8,104,188,100,40]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '3401'.))
[2025-09-06 16:41:21] error: 	zhc:tuya: Failed to sync time with '0xa4c138b442aa4702' (Error: ZCL command 0xa4c138b442aa4702/1 manuSpecificTuya.mcuSyncTime({"payloadSize":8,"payload":[104,188,72,12,104,188,100,44]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '33556'.))
[2025-09-06 16:41:25] error: 	zhc:tuya: Failed to sync time with '0xa4c1382913b655dd' (Error: ZCL command 0xa4c1382913b655dd/1 manuSpecificTuya.mcuSyncTime({"payloadSize":8,"payload":[104,188,72,17,104,188,100,49]},
```

I would be very grateful if you could bring the fix to those devices.
I can provide further logs if needed.
All big thumbs up for your great work @Koenkk !
J.

Koenkk, 2025-09-09:
@PeetersJan are you on the latest dev branch?

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

PeetersJan, 2025-09-09:
My bad !
Now I am and everything is peach!

[image]

true9741, 2025-09-15:
since moving to version [2.6.1] commit: [e9889e5d20e8911550d5231bc0349567cc2bb9ce] I am having failed to sync time in the Device type: TS0601 Zigbee Manufacturer: _TZE204_mrffaamu

[image]

I have taken the following debug information of one of my devices:

[2025-09-15 12:12:20] debug:    zh:controller:endpoint: Error: ZCL command 0x943469fffe12cf51/1 manuSpecificTuya.dataQuery({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 11543 - 1 - 228 - 61184>
    at Object.start (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/utils/waitress.ts:68:23)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:659:47)
    at ZStackAdapter.sendZclFrameToEndpointInternal (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:691:28)
    at /app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:439:20
    at Queue.execute (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZStackAdapter.sendZclFrameToEndpoint (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:437:16)
    at Request.send (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/controller/helpers/request.ts:86:25)
    at Endpoint.sendRequest (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:358:20)
    at Endpoint.zclCommand (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:1048:28)
    at Endpoint.command (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:811:23)
[2025-09-15 12:12:20] error:    zhc:tuya: Failed to query '0x943469fffe12cf51' on interval (Error: ZCL command 0x943469fffe12cf51/1 manuSpecificTuya.dataQuery({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) faile>

Thanks for your help.

tonioni, 2025-09-24:
Previously quickly mentioned ME201WZ still causes mcuVersionRequest errors in dev version (and makes whole network almost unusable), adding "respondToMcuVersionResponse: false" to tuya.js ME201WZ section fixes it.

0xa4c13899aed50edf = ME201WZ (_TZE284_kyyu8rbj)

data/log/2025-09-24.09-59-18/log.log:[2025-09-24 10:00:19] error:       zhc:tuya: Failed respond to version response '0xa4c13899aed50edf' (Error: ZCL command 0xa4c13899aed50edf/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":22206,"apsFrame":{"profileId":260,"clusterId":61184,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":192},"zclSequence":77,"commandIdentifier":11} timed out after 10000ms))

Koenkk, 2025-09-24:
@tonioni disabled it for the _TZE284_kyyu8rbj

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.