Koenkk/zigbee2mqtt #28465: Hive SLR1b temperature_setpoint_hold_duration being set to "null" rather than default 65535 after 2.6.1 update

mildmay48a, 2025-09-02:
### What happened?

Upgraded to 2.6.1-1 from 2.6.0-1 successfully. However I've noticed that temperature_setpoint_hold_duration  which has always defaulted to 65535 was now showing as "null". Each attempt to set the value to a numeric was successful but it was then reset to "null" some seconds later. 
Here are the log lines immediately after the upgrade & zigbee2mqtt restart

[2025-09-02 09:01:38] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/hive_thermostat', payload '{"last_seen":"2024-08-18T21:41:49.063Z","linkquality":105,"local_temperature":22.23,"occupied_heating_setpoint":19,"running_state":"idle","system_mode":"heat","temperature_setpoint_hold":true,"temperature_setpoint_hold_duration":65535}'
[2025-09-02 09:01:40] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/hive_thermostat', payload '{"last_seen":"2024-08-18T21:41:49.063Z","linkquality":102,"local_temperature":22.23,"occupied_heating_setpoint":19,"running_state":"idle","system_mode":"heat","temperature_setpoint_hold":true,"temperature_setpoint_hold_duration":null}'

### What did you expect to happen?

I expected that the value 65535 would remain as the default as "null" is not a numeric value.

I note that there were changes to hive.ts in this release but they look like they were for a new (old) Hive receiver.

### How to reproduce it (minimal and precise)

You'll need a Hive SLR1b. Once device is active - under (2.6.0) goto Devices under Zigbee2mqtt and use "Exposes" to see the value of temperature_setpoint_hold_duration. Upgrade to (2.6.1) & restart Zigbee2mqtt addon and repeat the step above. 
Then try setting the value of  temperature_setpoint_hold_duration to any numeric. Now watch in the Activity(LogBook) window and see the message temperature_setpoint_hold_duration became unknown
Check the value in the zigbee2mqtt/Devices/Exposes and the value given will be "null".

### Zigbee2MQTT version

2.6.1-1

### Adapter firmware version

20230507

### Adapter

Sonoff_Zigbee_3.0_USB_Dongle_Plus

### Setup

Add-on on Home Assistant on Raspberry Pi 5

### Device `database.db` entry

{"id":3,"type":"Router","ieeeAddr":"0x001e5e09023e0b11","nwkAddr":9461,"manufId":4153,"manufName":"Computime","powerSource":"Mains (single phase)","modelId":"SLR1b","epList":[5,6,7,8],"endpoints":{"5":{"profId":260,"epId":5,"devId":769,"inClusterList":[0,3,9,10,513,64768],"outClusterList":[0,10,25,1026,64768],"clusters":{"genBasic":{"attributes":{"zclVersion":1,"modelId":"SLR1b","manufacturerName":"Computime","powerSource":1,"appVersion":33,"stackVersion":2,"hwVersion":1,"dateCode":"20180504"}},"hvacThermostat":{"attributes":{"localTemp":2387,"systemMode":4,"occupiedHeatingSetpoint":2000,"tempSetpointHold":1,"tempSetpointHoldDuration":null,"runningState":0}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc1216","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc1216","endpointID":1},{"cluster":9,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc1216","endpointID":1},{"cluster":10,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc1216","endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc1216","endpointID":1}],"configuredReportings":[{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":41,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":513,"attrId":28,"minRepIntval":10,"maxRepIntval":3600,"repChange":null,"manufacturerCode":null},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":35,"minRepIntval":0,"maxRepIntval":3600,"repChange":0,"manufacturerCode":null},{"cluster":513,"attrId":36,"minRepIntval":0,"maxRepIntval":3600,"repChange":10,"manufacturerCode":null},{"cluster":513,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":41,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":513,"attrId":28,"minRepIntval":10,"maxRepIntval":3600,"repChange":null},{"cluster":513,"attrId":18,"minRepIntval":0,"maxRepIntval":3600,"repChange":10},{"cluster":513,"attrId":35,"minRepIntval":0,"maxRepIntval":3600,"repChange":0},{"cluster":513,"attrId":36,"minRepIntval":0,"maxRepIntval":3600,"repChange":10}],"meta":{}},"6":{"profId":260,"epId":6,"devId":769,"inClusterList":[513],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"7":{"profId":260,"epId":7,"devId":769,"inClusterList":[1026],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":769,"inClusterList":[1026],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":33,"stackVersion":2,"hwVersion":1,"dateCode":"20180504","zclVersion":1,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":190139088}

Koenkk, 2025-09-03:
> The value has remained stable at 65535 as expected/required.

Why is this expected/required?

mildmay48a, 2025-09-03:
Because that's the way it is expected to work according to https://www.zigbee2mqtt.io/devices/SLR1b.html
I use my HIVE system locally (no need for a Hive hub), and automations are based utilising mqtt for controlling heating schedules. I utilise native boosts when required and I use the value of temeperature_setpoint_hold_duration in the logic.
I'm sure I could come up with some workaround, but I've raised this as a problem due to the upgrade having changed the
way this works from the previous release.