Koenkk/zigbee2mqtt #28467: ZG-205Z does not change the value of the entity "presence" after updating to version 2.6.1-1

Masterrel, 2025-09-02:
### What happened?

ZG-205Z does not change the value of the entity "presence" after updating to version 2.6.1-1

### Zigbee2MQTT version

2.6.1-1

### Adapter firmware version

core: v2.9.8 / zigbee: -1

### Adapter

SLZB-06

### Setup

no

DadasupR, 2025-09-03:
Same issue, and LQI update is spamming the log, presence & presence state detection are null after a reset, force delete & repairing :

`info 2025-09-03 09:43:35z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:37.680Z","linkquality":70,"move_detection_max":7,"move_detection_min":0.2,"move_sensitivity":0,"none_delay_time":5,"presence":null,"presence_state":null,"small_move_detection_max":5.5,"small_move_detection_min":0.2,"small_move_sensitivity":9,"target_distance":5.88}'
info 2025-09-03 09:43:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:37.844Z","linkquality":76,"move_detection_max":7,"move_detection_min":0.2,"move_sensitivity":0,"none_delay_time":5,"presence":null,"presence_state":null,"small_move_detection_max":5.5,"small_move_detection_min":0.2,"small_move_sensitivity":9,"target_distance":5.88}'
info 2025-09-03 09:43:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:37.881Z","linkquality":76,"move_detection_max":7,"move_detection_min":0.2,"move_sensitivity":0,"none_delay_time":5,"presence":null,"presence_state":null,"small_move_detection_max":5.5,"small_move_detection_min":0.2,"small_move_sensitivity":9,"target_distance":5.88}'
info 2025-09-03 09:43:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:37.989Z","linkquality":81,"move_detection_max":7,"move_detection_min":0.2,"move_sensitivity":0,"none_delay_time":5,"presence":null,"presence_state":null,"small_move_detection_max":5.5,"small_move_detection_min":0.2,"small_move_sensitivity":9,"target_distance":5.88}'
info 2025-09-03 09:43:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:38.076Z","linkquality":76,"move_detection_max":7,"move_detection_min":0.2,"move_sensitivity":0,"none_delay_time":5,"presence":null,"presence_state":null,"small_move_detection_max":5.5,"small_move_detection_min":0.2,"small_move_sensitivity":9,"target_distance":5.88}'
info 2025-09-03 09:43:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:38.162Z","linkquality":70,"move_detection_max":7,"move_detection_min":0.2,"move_sensitivity":0,"none_delay_time":5,"presence":null,"presence_state":null,"small_move_detection_max":5.5,"small_move_detection_min":0.2,"small_move_sensitivity":9,"target_distance":5.88}'
info 2025-09-03 09:43:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:38.179Z","linkquality":76,"move_detection_max":7,"move_detection_min":0.2,"move_sensitivity":0,"none_delay_time":5,"presence":null,"presence_state":null,"small_move_detection_max":5.5,"small_move_detection_min":0.2,"small_move_sensitivity":9,"target_distance":5.88}'
info 2025-09-03 09:43:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:38.303Z","linkquality":76,"move_detection_max":7,"move_detection_min":0.2,"move_sensitivity":0,"none_delay_time":5,"presence":null,"presence_state":null,"small_move_detection_max":5.5,"small_move_detection_min":0.2,"small_move_sensitivity":9,"target_distance":5.88}'
info 2025-09-03 09:43:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:38.359Z","linkquality":76,"move_detection_max":7,"move_detection_min":0.2,"move_sensitivity":0,"none_delay_time":5,"presence":null,"presence_state":null,"small_move_detection_max":5.5,"small_move_detection_min":0.2,"small_move_sensitivity":9,"target_distance":5.88}'
info 2025-09-03 09:43:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:38.433Z","linkquality":76,"move_detection_max":7,"move_detection_min":0.2,"move_sensitivity":0,"none_delay_time":5,"presence":null,"presence_state":null,"small_move_detection_max":5.5,"small_move_detection_min":0.2,"small_move_sensitivity":9,"target_distance":5.88}'
info 2025-09-03 09:43:36z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:38.496Z","linkqualit [... cut ...]
info 2025-09-03 09:43:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload '{"breath_detection_max":5.5,"breath_detection_min":0.2,"breath_sensitivity":9,"illuminance":0,"indicator":"ON","last_seen":"2025-09-03T07:43:39.062Z","linkquality":70,"move_detection_max":7,"move_detection_min":0.2,"move_sensitivity":0,"none_delay_time":5,"presence":null,"presence_state":null,"small_move_detection_max":5.5,"small_move_detection_min":0.2,"small_move_sensitivity":9,"target_distance":5.88}'
info 2025-09-03 09:43:37z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/human presence detector', payload `

ayounas, 2025-09-03:
Same issue,

**Summary**
After updating to Zigbee2MQTT 2.6.1-1, my Tuya ZG-205Z (mmWave radar) stopped updating the boolean presence. The device still publishes target_distance regularly, but both presence and presence_state remain null. Rolling back to 2.6.0 immediately restores correct behaviour.

**What changed / Timeline**

At the scheduled nightly update, Home Assistant showed the entity going unavailable → none/unknown, and from then on presence/presence_state never recovered on 2.6.1. Downgrading Z2M to 2.6.0 brought presence back immediately with no other changes.

**Troubleshooting performed**

- Factory-reset + re-join several times.
- bridge/request/device/interview and .../configure.
- Force remove device + cleared retained MQTT state for its topics.
- Restarted Z2M/HA; network params unchanged (channel/PAN/network key).
- Enabled deep debug for zigbee-herdsman(-converters):tuya.

**Raw Tuya EF00 datapoints (debug logs)**
<details> <summary>Log excerpts (DPs seen while moving in front of the radar)</summary>

# Presence-state enum is being reported on DP=1 as different values (2 and 4):
... type 'commandDataReport', cluster 'manuSpecificTuya',
    data '{"dpValues":[{"datatype":4,"dp":1,"data":{"data":[2]}}]}'
... type 'commandDataReport', cluster 'manuSpecificTuya',
    data '{"dpValues":[{"datatype":4,"dp":1,"data":{"data":[4]}}]}'

# Distance is reported on DP=101 (e.g., 112 -> 1.12 m, 224 -> 2.24 m):
... data '{"dpValues":[{"datatype":2,"dp":101,"data":{"data":[0,0,0,112]}}]}'
... data '{"dpValues":[{"datatype":2,"dp":101,"data":{"data":[0,0,0,224]}}]}'

# But Z2M publishes presence fields as null:
MQTT publish: zigbee2mqtt/0xa4c1380f8ab06118
{"illuminance":0,"indicator":"ON","move_detection_min":0.3,"none_delay_time":30,
 "presence":null,"presence_state":null,"target_distance":1.12}
</details>

Interpretation: the device does report presence state on DP=1 (enum) and distance on DP=101, but on 2.6.1 the converter for this fingerprint does not map DP=1 → presence_state / presence. On 2.6.0, the same hardware maps correctly.

Bloodiss, 2025-09-03:
Same issue
Tuya ZG-205Z does not change the value of the entity "presence"

DebrodeD, 2025-09-03:
Same issue here - "presence" and "presence_state". They're just missing from the payload (where they used to be present before):

z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Motion Gym Gym Presence Sensor', payload '{"breath_detection_max":4,"breath_detection_min":1,"breath_ sensitivity":9,"illuminance":1096,"indicator":"ON","linkqual ity":192,"move_detection_max":4,"move_detection_min":1,"move _sensitivity":7,"none_delay_time":0,"small_move_detection_ma x":4,"small_move_detection_min":1,"small_move_sensitivity":7 ,"target_distance":3.92}'

closed 2025-09-03 by Koenkk