Koenkk/zigbee2mqtt #28468: After Update to 2.6.1 in Home Assistant device TS0044 changed model to another

NetSecond, 2025-09-02:
### What happened?

Hi,

today I have updated Zigbee2MQTT in Home Assistant from version 2.6.0 to 2.6.1.
After that, I had to delete the 4way switch and reconfigure it.

The main component is a Tuya TS0044 but in a version from manufacturer MOES.

Before Update:
Wireless switch with 4 buttons
Moes ZT-SY-EU-G-4S-WH-MS
https://www.zigbee2mqtt.io/devices/TS0044.html

After Update:
4 button portable remote control
Moes XH-SY-04Z
(https://www.zigbee2mqtt.io/devices/XH-SY-04Z.html

I had to reconfigure the device, blueprints and automations to get it work again but now in the frontend is the wrong device listet.

Best regards
Net

### What did you expect to happen?

Something went wrong in the device database in update 2.6.1.

### How to reproduce it (minimal and precise)

I have only updated from 2.6.0 to 2.6.1.

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

EmberZNet  8.0.2 [GA]

### Adapter

SONOFF Zigbee 3.0 USB-Dongle ZBDongle-E

### Setup

AddOn in Home Assistant OS (original image) on a Raspberry Pi 4B.

Koenkk, 2025-09-03:
`_TZ3000_kfu8zapd` should be detected as https://www.zigbee2mqtt.io/devices/TS0044.html in 2.6.1, if this is not the case, can you share a screenshot of the device about tab?

NetSecond, 2025-09-03:
[image]

Koenkk, 2025-09-04:
Seems they use the same manufacturer name for multiple devices (`_TZ3000_zgyzgdua` see https://github.com/Koenkk/zigbee2mqtt/issues/28224). So we cannot detect both properly (luckily only a cosmetic issue)

NetSecond, 2025-09-04:
0x7cc6b6fffe965a85 <= 4way button
0xa4c13805df973f65 <= 4way remote switch

NetSecond, 2025-11-15:
I found a solution (for me):
In device settings, delete the icon path.
After that, the correct image is shown again.
I found that at adding a second TS0044 from MOES.

closed 2025-11-15 by NetSecond