Koenkk/zigbee2mqtt #28477: [Feature request]: Contact sensor pairs as humidity sensor after Z2M update

doming52, 2025-09-03:
### Is your feature request related to a problem? Please describe

Tuya zigbee door sensor pairs as a humidity sensor after Z2M update:

Device type-EndDevice
Zigbee Model-TS0203
Zigbee Manufacturer-Zbeacon
Description-Temperature & humidity sensor
Support status-Supported
Firmware ID-0122052017
Manufacturer-[iHseno]
Model-[ZC-W1]
 

### Describe the solution you'd like

Please, Fix the update issue so it is recognized as a contact sensor again

### Describe alternatives you've considered

I tried to force an alternate pairing with no success

### Additional context

None

tbckr, 2025-09-03:
same issue here

Koenkk, 2025-09-03:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/28167#issuecomment-3246439100