Koenkk/zigbee2mqtt #28482: [BUG]: ZG-205Z (_TZE204_dapwryy7) presence state becomes 'unknown' after updating to 2.6.1-1

kvirtu1996, 2025-09-03:
### What happened?

### Description

After updating Zigbee2MQTT from version **2.6.0-1** to **2.6.1-1**, my **Tuya ZG-205Z** presence sensor (manufacturer _TZE204_dapwryy7) stopped correctly reporting the `presence` state in Home Assistant.

The `binary_sensor.kabinet_rux_presence` entity consistently shows `unknown`. Other values, such as `target_distance` and `illuminance`, are working without issues.

**Crucially, the problem disappeared after reverting back to version 2.6.0-1**. This confirms that the bug was introduced in version 2.6.1-1.

### Expected behavior

The sensor should correctly switch the state of `binary_sensor.kabinet_rux_presence` between `on` and `off` upon detecting presence. This functionality worked correctly on the previous version.

### Steps to reproduce

1. Update Zigbee2MQTT from version **2.6.0-1** to **2.6.1-1**.
2. Ensure the **Tuya ZG-205Z** device (model **_TZE204_dapwryy7**) is connected.
3. Check the state of the `binary_sensor.kabinet_rux_presence` entity in Home Assistant, which will remain `unknown`.

### Device and environment information

* **Zigbee2MQTT version:** `2.6.1-1` (the version where the issue occurs)
* **Device Model:** `ZG-205Z`
* **Device Vendor:** `Tuya`
* **Device Description:** `5.8 GHz human presence sensor`
* **Manufacturer:** `_TZE204_dapwryy7`

### Log output

### What did you expect to happen?

### Current device parameters

These parameters are being correctly read, confirming the device is alive:
* `binary_sensor.kabinet_rux_presence`: `unknown`
* `sensor.kabinet_rux_target_distance`: `2.24m`
* `sensor.kabinet_rux_illuminance`: `29lx`

### Zigbee2MQTT version

2.6.1-1

### Adapter firmware version

2025.08.3

### Adapter

SLZB-MR1

### Setup

1

kvirtu1996, 2025-09-03:
sorry, maybe it's written technically incorrectly, this is my first post and gemini helped write it

Deniallugo, 2025-09-04:
Same with 

Zigbee Model TS0601
Zigbee Manufacturer _TZE204_dapwryy7

tomkentpayne, 2025-09-04:
> The `binary_sensor.kabinet_rux_presence` entity consistently shows` unknown`

Is there now a `binary_sensor.kabinet_rux_occupancy` with 2.6.1-1?

I imagine this issue is related to [this pull request](https://github.com/Koenkk/zigbee2mqtt/pull/28329), which corrected the semantics expected by Home Assistant around the term "presence" vs. "occupancy".

Maheidem, 2025-09-04:
Im getting my whole stack unavailable every few minutes since this update. As if it was unplugging the dongle from the server

ahorobets, 2025-09-04:
the same issue _TZE204_dapwryy7 (5.8 GHz human presence sensor)